realsenseai/librealsense #5895: Android Stream Index

erikgallegos, 2020-02-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Android 8.0  | 
| Platform                                 | Galaxy S8 |
| SDK Version                            |  2.31.0                         | 
| Language                            |  kotlin                       | 
| Segment			| Smartphone    | 

### Issue Description
I am currently setting up my stream like this: 

```
Config().use { config ->
      config.enableStream(StreamType.DEPTH, STREAM_WIDTH, STREAM_HEIGHT)
      config.enableStream(StreamType.COLOR, STREAM_WIDTH, STREAM_HEIGHT)
      mPipeline.start(config)
}
```
The issue is that sometimes the depth stream shows up first and sometimes the color stream shows up first. So I tried using another `enableStream` function which takes in an index, which I am assuming is to set the stream order.

I am now setting up my stream like this:

```
Config().use { config ->
      config.enableStream(StreamType.COLOR, 0, STREAM_WIDTH, STREAM_HEIGHT, StreamFormat.ANY, 0)
      config.enableStream(StreamType.DEPTH, 1, STREAM_WIDTH, STREAM_HEIGHT, StreamFormat.ANY, 0)
      mPipeline.start(config)
}
```

But now I do not see either stream on the screen. 

Is this the proper way to use these functions? Is there a way to guarantee the order in which the streams are displayed?

Thank you.

kafan1986, 2020-02-22:
Last parameter is FPS which you have declared as 0. If the FPS you are looking for is 15 then it should be.

config.enableStream(StreamType.COLOR, 0, STREAM_WIDTH, STREAM_HEIGHT, StreamFormat.RGB8, 15)
      config.enableStream(StreamType.DEPTH, 0, STREAM_WIDTH, STREAM_HEIGHT, StreamFormat.Z16, 15)

Also, the correct stream format is also needs to be defined, depending on which format you want the data to be in.

erikgallegos, 2020-02-24:
@kafan1986 
`config.enableStream(StreamType.DEPTH, STREAM_WIDTH, STREAM_HEIGHT)` internally calls 
`config.enableStream(StreamType.DEPTH, -1, STREAM_WIDTH, STREAM_HEIGHT, StreamFormat.ANY, 0)` so I don' think that is the case

dorodnic, 2020-02-25:
Hi @erikgallegos 
This is not what stream index does. There is no way to select which stream will start first as far as I know.

erikgallegos, 2020-02-28:
@dorodnic Just to be clear, I don't necessarily want to select which stream will start first, just the on-screen order of the streams. For example, sometimes the depth stream will be at the top of the screen and the color stream will be underneath it, and sometimes their positions are flipped.

But you are saying this is not possible to control right?

dorodnic, 2020-03-03:
I think the way to approach this is to modify the application. Right now UI layout is picked somewhat at random (maybe based on which one arrives first, or maybe based on something else).

closed 2020-03-03 by erikgallegos