realsenseai/librealsense #5896: Announcement:  Highlighting the 'JetScan' 3D scanner build project for Jetson Nano

MartyG-RealSense, 2020-02-22:
Hi everyone,

As highlighted by the RealSense Twitter account, JetScan is a fully documented build project that describes itself as: "smart and fast GPU 3D scanner with Jetson Nano and Intel depth sensor. 
 Metrically accurate RGBD 3D scanner and instant 3D reconstruction".

https://www.hackster.io/devshank/jetscan-16a521

devshank3, 2020-02-23:
Thanks @MartyG-RealSense for featuring my project 

https://twitter.com/ShankarHaran3/status/1231456562927087617?s=20

MartyG-RealSense, 2020-02-23:
you are very welcome!   It's great when RealSense community members document their projects thoroughly with guides, papers or long videos so that others can learn from it for their own projects.  Thanks!

closed 2020-02-25 by MartyG-RealSense