realsenseai/librealsense #5897: MacOS build failing

militaryCoder, 2020-02-23:
While building library via CMake under MacOS I get a problem during building include file `./include/librealsense2/rsutil.h`.
I get "ambiguous call"-error:
``` cpp
      rsutil.h:96:17: error: 
      call to 'abs' is ambiguous
            if (std::abs(f) < FLT_EPSILON)
                ^~~~~~~~
/usr/include/stdlib.h:129:6: note: candidate function
int      abs(int) __pure2;
         ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstdlib:167:44: note: 
      candidate function
inline _LIBCPP_INLINE_VISIBILITY long      abs(     long __x) _NOEXCEPT {return  labs(__x);}
                                           ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstdlib:169:44: note: 
      candidate function
inline _LIBCPP_INLINE_VISIBILITY long long abs(long long __x) _NOEXCEPT {return llabs(__x);}
```
Also I've got few errors in `./common/viewer.cpp`:
```cpp
viewer.cpp:1694:26: error:
      non-constant-expression cannot be narrowed from type 'double' to 'float' in initializer list [-Wc++11-narrowing]
            { 0, cos(x), -sin(x), 0 },
```
I get 8 errors of this type in this file.

Maybe I forget to use some compiler flags or something like that but it would be nice if you'll investigate it yourself as you are the authors of the code.

militaryCoder, 2020-02-23:
I've figured out how to fix it.
Earlier I've added `std::` prefix to `abs(f)` call but `stdlib.h` is included so it was searching for `abs()` definition in C++ libraries. I just removed it and this error turned into warning.
But there are still errors with casts in initializer lists.

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

Could you share your build environment info?
which compiler do you use?

militaryCoder, 2020-03-04:
@RealSenseCustomerSupport, I'm using C/C++ compiler `Apple CLang 8.0.0.8000042` and compiling under **macOS 10.12.6**.

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

I can't reproduce this issue using the Xcode 11.3.1.
Would you upgrade you compiler?
BTW, Please double check the "c++11" option in project file has been enabled.

militaryCoder, 2020-03-05:
@RealSenseCustomerSupport, I cannot upgrade XCode and compiler because my notebook is too old for upgrading and do not support newer macOS versions.

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

There is a not very good idea before editing the code.
Would you like to install GCC? :)

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

@militaryCoder Did you get chance to install GCC to try? Looking forward to your update. Thanks!

militaryCoder, 2020-03-31:
@RealSenseCustomerSupport, I've got a couple of free days so I'll try it as soon as possible. Thanks for the reminder.

neilyoung, 2020-03-31:
I'm not having issues compiling on Mac. What is your CMake command line?

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@militaryCoder Any update? Thanks!

militaryCoder, 2020-04-10:
@RealSenseCustomerSupport, I'm currently working on it but my laptop is too laggy lately so it takes much more time than it really needs.

militaryCoder, 2020-04-10:
That's what I mean. :)

[image]

militaryCoder, 2020-04-10:
Well, while building under **g++ 4.2.1** I've got similar error:

> Also I've got few errors in `./common/viewer.cpp`:
> 
> ```c++
> viewer.cpp:1694:26: error:
>       non-constant-expression cannot be narrowed from type 'double' to 'float' in initializer list [-Wc++11-narrowing]
>             { 0, cos(x), -sin(x), 0 },
> ```
> 
> I get 8 errors of this type in this file.

RealSenseCustomerSupport, 2020-04-16:
----------------------------------------------

at least GCC 5 … not the one in Xcode.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Please make sure you have used GCC which you have installed.

militaryCoder, 2020-04-16:
Well, I'm trying to change default compiler on my Mac, but still no luck.
Do you know any good tutorial how to use compiler not included in XCode toolchain on Mac?
Could use some help with it.

RealSenseCustomerSupport, 2020-05-06:
----------------------------------------------

Please try the commands like this. There work in my laptop (macOS 10.15)
brew install pkg-config libusb glfw gcc
cmake -DCMAKE_C_COMPILER=/usr/local/bin/gcc-9 -DCMAKE_CXX_COMPILER=/usr/local/bin/g++-9 ..

Reference:)
https://stackoverflow.com/questions/24380456

militaryCoder, 2020-05-06:
@RealSenseCustomerSupport, I've already found it out but had no time to try it out. Will check it this evening.

RealSenseCustomerSupport, 2020-05-11:
----------------------------------------------

@militaryCoder How is it going? Any update from your side? Thanks!

militaryCoder, 2020-05-11:
@RealSenseCustomerSupport, I've built project under g++ 9.3.0 and it finally was built. Thanks for your patience.

closed 2020-05-11 by militaryCoder