realsenseai/librealsense #5898: D435 with two T265

stino78, 2020-02-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435,T265} | 
| Operating System & Version |   {Linux (Ubuntu)   | 
| Kernel Version (Linux Only)    |  {5.3.0-28-generic}                                         | 
| Platform                                 | {PC}  |
| SDK Version                            |  {legacy / 2.32.1299 and 2.32.1331 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {0thers }                          | 

### Issue Description
<
Hi guys, I need some serious help.

I am trying to connect one D435 depth camera together with two T265 on the same machine, and I have hard time doing it. The machine is a Ubuntu 18.04 and I am using the precompiled SDK with the Python bindings. I am also trying the same code on Win10.

I want to open the D435 in a thred and the two T265 in a separate thread.

When using the legacy version (2.32.1299), once depth camera is started, trackers are not available in a separate thread error when calling context.devices is 'Unable to open device interface'. This error is well documented by other Git issues.

But when upgrading to 2.32.1331, even though trackers are now available in the separate thread, the input from the D435, both depth and RGB, become laggy and skips many frames.

Any ideas? 

Will this be fixed with the upcoming release? In this case, when will it be released?

Thank you,

Agostino
>

giuliafoca, 2020-02-23:
I'm having almost the same problem: I want to run the D435 on one thread and T265 from python. I'm working on windows 10 with SDK 2.32.1331 and the color and depth streams from the D435 realsense camera have frequent pauses of about half a second.

Can you help me with that?

Thank you 
Giulia

vdutell, 2020-02-23:
Same issue & same platform - but noticed also that this lag in the RGB stream is present even in realsense viewer with the new version (2.32.1331).  Lag is present in the following configurations:

Depth only: Sometimes Lag
IR only: No Lag
Depth + IR: Always Lag in both streams
RGB only: Sometimes Lag
RGB + depth: Always Lag in both streams
RGB + dpeth + IR: Always Lag in all streams

Thanks for your help,
Vasha

stino78, 2020-02-24:
Hi vdutell,

I made the same tests, and I have the same results! Also the Realsense Viewer is crashing!

stino78, 2020-02-26:
Any insights from anyone?

closed 2024-04-01 by MartyG-RealSense