realsenseai/librealsense #5899: C# Wrapper Export to Ply function does not seem to save texture data

maxccgh, 2020-02-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-03-03:
Apologies for the delay in responding.

Have you used **pc.map_to(color)** to map the points to the color frame with a 'uvmap' (texture coordinates) please?

This subject is a bit complicated when programming in C#.  Originally the point cloud texturing code would have been something like this:

*************

auto color = frames.get_color_frame();
points = pc.calculate(depth);
pc.map_to(color);
points = pc.calculate(depth);

***********

The PointCloud.Calculate method was made obsolete though, and may generate a harmless error warning such as this: 

**PointCloud.Calculate method is obsolete ("This method is obsolete. Use Process method instead")**

It was suggested instead to do C# point cloud calculation with a processing block instead, such as in the example program here:

https://github.com/IntelRealSense/librealsense/blob/development/wrappers/csharp/Documentation/cookbook.md#generics-support

MartyG-RealSense, 2020-03-12:
This case will be closed after 7 days from now if there are no further questions.   Thanks!

closed 2020-03-19 by MartyG-RealSense