realsenseai/librealsense #5901: HID set_power 1 failed for D435i

studennis911988, 2020-02-24:
Hi, I'm using the latest SDK & D400 FW and I encounter this error message when using the IMU of D435i camera.
`WARNING [140323645359872] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.5/0003:8086:0B3A.0004/HID-SENSOR-200076.2.auto/iio:device1/buffer/enable
`
The weird thing is that, the problem disappear **every time** I re-plug the USB cable, but the error pop out again in the next round. Since it's quite annoying, I looking for your help.

Thanks in advanced.

The full log file is underneath.
> [0m[ INFO] [1582543338.623086993]: RealSense ROS v2.2.12[0m
[0m[ INFO] [1582543338.623110300]: Running with LibRealSense v2.31.0[0m
[0m[ INFO] [1582543338.954458334]:  [0m
[0m[ INFO] [1582543338.971479264]: Device with serial number 848312070696 was found.
[0m
[0m[ INFO] [1582543338.971580971]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0 was found.[0m
[0m[ INFO] [1582543338.971923793]: Device with name Intel RealSense D435I was found.[0m
[0m[ INFO] [1582543338.980589410]: Device with port number 2-3 was found.[0m
[0m[ INFO] [1582543338.988989968]: getParameters...[0m
[0m[ INFO] [1582543339.042834525]: setupDevice...[0m
[0m[ INFO] [1582543339.042853017]: JSON file is not provided[0m
[0m[ INFO] [1582543339.042859499]: ROS Node Namespace: camera[0m
[0m[ INFO] [1582543339.042872171]: Device Name: Intel RealSense D435I[0m
[0m[ INFO] [1582543339.042882181]: Device Serial No: 848312070696[0m
[0m[ INFO] [1582543339.042891896]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0[0m
[0m[ INFO] [1582543339.042904042]: Device FW version: 05.12.02.100[0m
[0m[ INFO] [1582543339.042915035]: Device Product ID: 0x0B3A[0m
[0m[ INFO] [1582543339.042922777]: Enable PointCloud: Off[0m
[0m[ INFO] [1582543339.042931930]: Align Depth: Off[0m
[0m[ INFO] [1582543339.042943330]: Sync Mode: On[0m
[0m[ INFO] [1582543339.043199281]: Device Sensors: [0m
[0m[ INFO] [1582543339.043389606]: Stereo Module was found.[0m
[0m[ INFO] [1582543339.043431284]: RGB Camera was found.[0m
[0m[ INFO] [1582543339.043447990]: Motion Module was found.[0m
[0m[ INFO] [1582543339.043487497]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...[0m
[0m[ INFO] [1582543339.043502106]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...[0m
[0m[ INFO] [1582543339.043514700]: (Pose, 0) sensor isn't supported by current device! -- Skipping...[0m
[0m[ INFO] [1582543339.043665770]: num_filters: 0[0m
[0m[ INFO] [1582543339.043676858]: Setting Dynamic reconfig parameters.[0m
[0m[ INFO] [1582543339.243775738]: Done Setting Dynamic reconfig parameters.[0m
[0m[ INFO] [1582543339.247775040]: depth stream is enabled - width: 640, height: 480, fps: 60, Format: Z16[0m
[0m[ INFO] [1582543339.248069330]: infra1 stream is enabled - width: 640, height: 480, fps: 60, Format: Y8[0m
[0m[ INFO] [1582543339.248265752]: setupPublishers...[0m
[0m[ INFO] [1582543339.251297325]: Expected frequency for depth = 60.00000[0m
[0m[ INFO] [1582543339.285196128]: Expected frequency for infra1 = 60.00000[0m
[0m[ INFO] [1582543339.303265916]: Start publisher IMU[0m
[0m[ INFO] [1582543339.304583384]: setupStreams...[0m
[0m[ INFO] [1582543339.305084646]: insert Depth to Stereo Module[0m
[0m[ INFO] [1582543339.305223087]: insert Infrared to Stereo Module[0m
[0m[ INFO] [1582543339.305255225]: insert Gyro to Motion Module[0m
[0m[ INFO] [1582543339.305275673]: insert Accel to Motion Module[0m
 24/02 19:22:19,307 WARNING [140323645359872] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.5/0003:8086:0B3A.0004/HID-SENSOR-200076.2.auto/iio:device1/buffer/enable
 24/02 19:22:19,307 WARNING [140323406952192] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.5/0003:8086:0B3A.0004/HID-SENSOR-200073.1.auto/iio:device0/buffer/enable

ev-mp, 2020-02-26:
@studennis911988 hello,
This error message is rooted in kernel initialization of the IMU sensor. The device loading and especially the power-management of the required kernel IIO modules is not well-defined, and may leads to incomplete node build-up/start-up.
SDK provides an effective work-around that rectifies and enforces sensor functionality regardless of kernel (mis)behavior. So unless you get this message along with actual IMU error (e.g. no IMU data produced), then this warning can be disregarded.

studennis911988, 2020-02-26:
@ev-mp 
Thanks for your advise, but it's tedious to plug and unplug the cable every time I start the program, so is there any solution?

ev-mp, 2020-02-26:
@studennis911988  - please clarify whether this is an actual issue with IMU  - 
Do you mean that you cannot restart ROS node without re-plugging the camera?

studennis911988, 2020-02-27:
@ev-mp Sorry for the misunderstanding.
The ROS node can start successfully without re-plugging the cable but with the HID error I mentioned, and by re-plugging the cable, everything goes fine without any error message. However, the error message pops out again in the next round when I launch the ROS node again.

ev-mp, 2020-02-27:
@studennis911988 , thanks for the confirmation. 
To reconfirm - this message is informative only and does not affect sensor operation. There is no reason to replug the camera when this message is fired.
We also don't want to suppress this message as this allows to trace kernel changes.

studennis911988, 2020-02-29:
@ev-mp 
Sadly, in my device,  when the warning message fires, either accelerator or gyroscope can't output any data (I can't remember which one didn't function well), causing it being quite tedious to re-plug the cable.
I'm wondering if there is any solution for it.

RealSenseSupport, 2020-05-29:
Hi @studennis911988 

With latest FW and librealsense release, are you still encountering this issue?

*If we don’t hear from you in 7 days, this issue will be closed.

studennis911988, 2020-05-30:
@RealSenseSupport Thank for reminder, I'll give it a shot today and report the outcomes.

RealSenseSupport, 2020-08-20:
Hi @studennis911988

With latest FW and librealsense release, are you still encountering this issue?  Did you get a chance to try it out?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-09 by RealSenseSupport