realsenseai/librealsense #5902: ros-melodic-librealsense2 dependencies should be the same as librealsense2 dependencies

boilerbots, 2020-02-24:
The ros-melodic-librealsense2 dependencies cause other undesirable packages to be pulled in to a robot including gcc and kernel header files. Please make the dependencies simple like the normal librealsense2 package.

**librealsense2**
Depends: libusb-1.0-0 (>= 2:1.0.16), libc6 (>= 2.27), libgcc1 (>= 1:3.0), libstdc++6 (>= 6)

**ros-melodic-librealsense2**
Depends: libc6 (>= 2.27), libgcc1 (>= 1:3.0), libstdc++6 (>= 6), libusb-1.0-0 (>= 2:1.0.16), dkms, libssl-dev, libusb-1.0-0-dev, linux-headers-generic, udev

It would also help simplify configuration management if you also created a **ros-melodic-librealsense2-dev** package so a project would not require the Intel packages and the ROS packages and they would be synchronized together.

RealSenseCustomerSupport, 2020-04-09:
----------------------------------------------

Hi @boilerbots,

We are reviewing your change proposal request and will keep you informed about any updates.

Thank you!

RealSenseSupport, 2020-10-29:
WIP. Please check PR https://github.com/IntelRealSense/librealsense/pull/7674

RealSenseSupport, 2020-11-19:
@boilerbots PR#https://github.com/IntelRealSense/librealsense/pull/7674 now merged to LRS v2.40 and released now. Could you please try LRS v2.40? Thanks!

RealSenseSupport, 2020-11-26:
@boilerbots Any other questions about this? Please note that this will be closed if we don't get response from you for another 7 days. Thanks!

closed 2020-12-07 by boilerbots