realsenseai/librealsense #5903: How to store all frames with timestamps? null pointer passed for argument "frame_ref"

yash505, 2020-02-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (5.11.4) | 
| Operating System & Version |   {Win (10) / Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
What is the best way to store the realsense frames at high frequency as and when they are streamed? The goal is to store the frame with its timestamp and associate a frame with the instantaneous pose of a robot.

I always get the error - 
RuntimeError: null pointer passed for argument "frame_ref"
This error doesn't show when I call `wait_for_frames()` instead. But using `wait_for_frames` is storing old frames and not the latest ones.

I am using a very naive method of appending to a list. What would be the best way to go about this? I am using the latest pyrealsense version 2.32.1.1331
```
def initRealSenseCamera(resolutionWidth=1280, resolutionHeight=720, frameRate=6):
    # Configure depth and color streams
    pipeline, mtx, align, config = 0,0,0,0
    while 1:
        print("Trying")
        try:
            pipeline = rs.pipeline()
            config = rs.config()
            cfg = pipeline.start(config)
            # config.enable_stream(rs.stream.depth, resolutionWidth, resolutionHeight, rs.format.z16, frameRate)
            config.enable_stream(rs.stream.color, resolutionWidth, resolutionHeight, rs.format.bgr8, frameRate)
            align_to = rs.stream.color
            align = rs.align(align_to)
            break       
        #print("Excepting")    
        except Exception as e:
            print("Camera not detected yet..")
            pass
        time.sleep(1)   
    return pipeline, mtx, align, config

pipeline, cameraMatrix, align, config = initRealSenseCamera()
frame_store = []
while True:
	frames = pipeline.poll_for_frames()
	frame_timestamp_ms = time.time_ns() / 1000000
	color_frame = frames.get_color_frame()
        if (color_frame.get_data()):
	        frame_store.append(color_frame)
```
Then retrieve that frame depending on closest timestamp

dorodnic, 2020-02-25:
Hi @yash505 
You need to call `frame.keep()` on any frame you are intending to store (keep) for longer than couple of frame periods. If you store a frameset, you also need to call `keep` on the color / depth frames within it. Without calling `keep` the SDK is limited to a fixed frame pool, once all frames are in use, you will either get old frames (from `wait_for_frames`) or `null` (from `poll_for_frames`)

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@yash505 Any other questions with dorodnic's reply? Looking forward to your update. Thanks!

yash505, 2020-03-06:
Thank you @dorodnic . I think I am good for now.

RealSenseCustomerSupport, 2020-03-09:
----------------------------------------------

@yash505 Glad to hear that. Close this accordingly. Thanks!

closed 2020-03-09 by RealSenseCustomerSupport