realsenseai/librealsense #5904: RealSense Viewer Crashes when opening bag file

ortego26, 2020-02-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi, I am currently using a d415 for some data collection that gets uploaded into dropbox, and then retrieved at a different computer. 

When I try to open the recorded .bag files, the realsense viewer software closes with no error message, it just closes and stops. Any idea why this could happen?

dorodnic, 2020-02-25:
Hi @ortego26 
ROS-bag file formatter is suffering from an issue where if the recording is stopped abruptly, the resulting bag file would not get indexed. I suspect this may be the case. 
I would also try copying the file to main hard-drive (as opposed to network drive)
One of the tools within the SDK is called rosbag-inspector and is accessing files using ROS-bag library directly (not via SDK API). It can also offer some information about the file

ortego26, 2020-02-26:
Hi @dorodnic 
That's what I was suspecting, but when the data is collected locally, it can be accessed no problem but then after upload and download in a different computer, even after moving the file to the hard drive, it seems to crash the viewer.

Here is a snippet of the code that I suspect is not stopping the recording abruptly.

# Get current time to name the file
        Date = time.asctime(time.localtime(time.time())).replace(':','-')
        file_path = bagfiles_dir_path_new + str(int(time.time())) + ' ' + Date + '.bag'
        
        # Configure depth and color streams
        pipeline = rs.pipeline()
        config = rs.config()
        config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, CAMERA_FRAMERATE)
        config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, CAMERA_FRAMERATE)
        config.enable_record_to_file(file_path)
        
        # Start streaming
        print("creating bag file "+file_path,flush=True)
        pipeline.start(config)
        t0 = time.time()
        
        try:
            while True:
                # Wait for a coherent pair of frames: depth and color
                frames = pipeline.wait_for_frames()
                depth_frame = frames.get_depth_frame()
                color_frame = frames.get_color_frame()
                if not depth_frame or not color_frame:
                    continue
                if (time.time() - t0) >= SECONDS_PER_RECORDING:
                    print("saving bag file "+file_path,flush=True)
                    break
        finally:
            # Stop streaming
            pipeline.stop()
            del(pipeline)
            del(config)

dorodnic, 2020-02-26:
That's very strange, perhaps permissions issue? Linux / Windows?

ortego26, 2020-03-04:
Hi @dorodnic. I found the issue. On the system that was working before (Windows) the version of the pyrealsense was 2.31.0.1235

When I set up a new system for data collection, I created a new environment that was using the pyrealsense2 version 2.33.1336

Reverting to the original fixed the bag files and they do not crash anymore. 

Maybe this was an issue with that update?

RealSenseCustomerSupport, 2020-06-11:
----------------------------------------------

Hi ortego26,

Will you be needing further help with this?

Are you still experiencing this on latest librealsense? As a general rule of thumb I would keep the versions of the SDK on the system that's recording the bag file and the one that's playing it the same.

If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-06-22 by RealSenseSupport