realsenseai/librealsense #5905: USB2 mode auto exposure issue with firmware v5.12.2.100

oor001, 2020-02-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.12.2.100 | 
| Operating System & Version |   {Win 10 / Ubuntu 18  | 
| Platform                                 | PC |
| SDK Version                            |  2.20.0                        | 
| Language                            |  C++                        | 

### Issue Description
We have found an issue with the auto exposure of the IR cameras on the D435 module when using the latest firmware and a USB 2 connection.

We previously encountered issues with auto exposure (see #4140, #5273 and https://support.intelrealsense.com/hc/en-us/community/posts/360033445713-D435-Sunlight-saturation-problem). The original issue is now fixed but it seems that as a result of fixing the previous issue of the camera under-exposing indoors, it is now overexposing, especially in brightly lit outdoor environments.

We have tested this with several cameras and firmware versions. There is no issue with USB 3 but unfortunately we have no option to use that. Here is an example of the issue:

[image]

Would it be possible to get some sort of explanation as to why this issue keeps flip-flopping and coming up over and over again? Additionally, is there information somewhere on why the USB3 and USB2 firmware behaviors differUSB2 mode auto exposure issue with firmware v so much? These ongoing problems are causing serious issues for our project.

agrunnet, 2020-02-25:
You are correct. The lowest exposure possible with USB2 is HIGHER than that possible with USB3. 
The root cause for this is that the D435 stereo sensors are extremely sensitive to light, so we are using a fractional exposure setting to make them work outdoors in bright sunlight. This fractional rate is dependent on the clock, which is slower on USB2 vs USB3. 
We do not expect to be able to fix this. The previous low-intensity on USB2 was actually with the imagers in a bad state so we do not want to revert to that. 
At this stage I would suggest using some neutral density filters to attenuate the light if you know you are going to be operating in bright sunlight. See our white paper on this. 
https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400
I know this is not what you wanted to hear, but it is currently the only solution.

agrunnet, 2020-02-25:
Actually looking at captured image, we noticed that you are running at 15fps. 
If you can, please try to run at 30fps. It should help 2x.

oor001, 2020-02-25:
Thanks for your response, I really appreciate the explanation regarding clock rates and USB2 vs USB 3. I will try again with 30FPS. I did notice the default speed seems to have dropped from 30 to 15 in the latest update but I didn't think to try that. Thanks also for your suggestion of filters.

RealSenseCustomerSupport, 2020-03-03:
----------------------------------------------

Hi oor001

Is the 30FPS helpful to your case? if not, you will have to use a ND filter to lower the sun light.
A ND4 filter should make the exposure equal to what it is under USB3 when you run camera at 30FPS under USB2.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

oor001, 2020-03-04:
Hi - I've tested 30FPS and the minimum exposure seems identical to 15. We have gone ahead and ordered some ND filters to try. Thanks, feel free to close this issue now.

closed 2020-03-05 by RealSenseCustomerSupport