realsenseai/librealsense #5906: Typo in frame_queue.m and mex error in destructor of frame_queue

sw290, 2020-02-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400, D415 } | 
| Firmware Version                   | (05.12.01.00) | 
| Operating System & Version |   {Win (10)}  | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.31.0> }                          | 
| Language                            |  {MATLAB }                          | 
| Segment			|  {Robot }                          | 

When calling:

frameQ = realsense.frame_queue(1);

Results in the matlab error:
'Error: File: frame_queue.m ... Illegal use of reserved keyword "end"'
However upon removal of this extra end in frame_queue.m results instead in:

Warning: The following error was caught while executing 'realsense.frame_queue' class destructor:
Error using librealsense_mex
Unknown Command received.
Error in realsense.frame_queue/delete (line 20)
                realsense.librealsense_mex('rs2::frame_queue', 'delete', this.objectHandle);

I've had a look at the example for using frame_queue for the python wrapper, but there is no option to pass 'frameQ' to pipeline.start (eg. pipeline.start( config, frameQ ) ).

1. Is this intentional or a bug?

2. Would someone be able to write a quick example of how to use the post processing filters (and framequeue) using the MATLAB wrapper?

Thank you for your time.

lramati, 2020-02-25:
These are things that are currently missing from the wrapper. I will add them in soon

RealSenseCustomerSupport, 2020-05-02:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

sw290, 2020-05-04:
Is there any ETA or roadmap on "soon" for adding the missing functionality in the MATLAB wrapper?

closed 2020-07-06 by dorodnic