realsenseai/librealsense #5907: D435i multi camera accel data issue

malith1992, 2020-02-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (8.1/10)   | 
| Platform                                 | PC  |
| Language                            |  {python }                          | 
| Segment			|  {Robot}                          | 

I want to use two D435i cameras and get accel data from both cameras using the following code. But sometimes it gives same accel data for both cameras. But sometimes it gives different data for each camera. It seems like the pipelines are messing with the device ID. But i can't figure it out.

```
import pyrealsense2 as rs
import numpy as np
from math import *
import math
import time

def getConnectedDevices():
    connect_devices = []
    context = rs.context()
    for d in context.devices:
        if d.get_info(rs.camera_info.name).lower() != 'platform camera':
            connect_devices.append(d.get_info(rs.camera_info.serial_number))
    connect_devices.sort()
    return connect_devices

def getAngle(connect_device):
    print(connect_device)
    anglepip1 = rs.pipeline()
    angleconfig1 = rs.config()
    angleconfig1.enable_device(connect_device)
    angleconfig1.enable_stream(rs.stream.accel)
    angleconfig1.enable_stream(rs.stream.gyro)
    anglepip1.start(angleconfig1)

    try:
        while True:
            f1 = anglepip1.wait_for_frames()
            accel1 = f1[0].as_motion_frame().get_motion_data()
            return accel1

    finally:
        anglepip1.stop()
        print("pipe stopped")

accel = []
connect_devices = []
connect_devices = getConnectedDevices()
devices = len(connect_devices)
print(connect_devices)
accel.insert(0,getAngle(connect_devices[0]))
accel.insert(1,getAngle(connect_devices[1]))
print(accel)
```

dorodnic, 2020-02-25:
Hi @malith1992 
We are in process of making a new release, that should have this resolved. 
You can install release candidate using test pypi or wait for formal release (or build from source using the development branch)

malith1992, 2020-02-25:
> Hi @malith1992
> We are in process of making a new release, that should have this resolved.
> You can install release candidate using test pypi or wait for formal release (or build from source using the development branch)

So as you say this is a bug in the library. So I have nothing to do to solve this issue right ?

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

Hi @malith1992,

Did you have a chance to install the release candidate/to build it from source? If you did, please provide us with the feedback.

Thank you!

RealSenseCustomerSupport, 2020-03-03:
----------------------------------------------

@malith1992,

Do you still need help with this question?

malith1992, 2020-03-03:
I had no time to install the release candidate. But actually the issue I have mentioned is not in the Linux environment. My actual code is running on Linux environment so I could move on. I use windows only for coding. Once I got time, I will try the release candidate. Until then I don't need help on this question. Thank you very much

closed 2020-03-03 by malith1992