realsenseai/librealsense #5908: How to rectify depth image so it can align with RGB image using .bag files?

jungseokhong, 2020-02-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 14/16/17)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/NVIDIA Jetson   |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I have .bag files and when I extracted images, I found that depth images and rgb images are from same frames but the ratio is different. I recorded using realsense viewer with 848*640 resolution for both depth and rgb. I thought I might be able to fix this by using depth_scale but I can't run "align-depth2color.py" in examples, using bag files.

https://github.com/IntelRealSense/librealsense/blob/a1f2961b3494075581dd680e76850027f3f02768/wrappers/python/examples/align-depth2color.py#L20
below line 2 I inserted following line
`rs.config.enable_device_from_file(config, 'mybag.bag')`.
config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 848, 480, rs.format.bgr8, 30)

However, I got following errors.
profile = pipeline.start(config)
RuntimeError: Couldn't resolve requests

I believe it is because the code is meant for streaming rather than reading from .bag files.
I want to generate rgb and depth images from this bag files so they are aligned pixel by pixel and have same sizes. For instance, pixel (10,10) in rgb should have same object in pixel (10,10) in depth.
Is there any way to handle this situation using bag files? If I run the above script when I record, will it guarantee that the rgb and depth images are identical in terms of pixel locations?

I am using pyrealsense wrapper for this and my model is D435i.

dorodnic, 2020-02-25:
Hi @jungseokhong 
You can inspect what was recorded inside the bag file using `rs-enumerate-devices -p filename.bag`. What was recorded is what you can play. If you need the two streams aligned, we provide `rs2::align` algorithm to spatially align a set of frames. This must be done explicitly since alignment is relatively costly operation and we don't want to run it without explicit command from the user. Hope this helps

jungseokhong, 2020-02-25:
Hi @dorodnic 
Thanks for the reply.

I tried the command and below is the output. It seems fine?
can you point me to the script to align them spatially and save them separately?
I can extract them using rs-convert without problems. but they are spatially mismatched. temporally aligned.

Device info: 
Playback Device (/home/mybag.bag)
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 
    Firmware Version              : 	05.12.02.100
    Recommended Firmware Version  : 	05.12.01.00
    Physical Port                 : 	2-1.3-3
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 
    Firmware Update Id            : 

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Depth	  848x480	@ 30Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  848x480	@ 30Hz	   RGB8

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format

dorodnic, 2020-02-26:
Hi @jungseokhong 
It should be something like:
```python
import pyrealsense2 as rs
p = rs.pipeline()
c = rs.config()
c.enable_devie_from_file(...)
p.start(c)
align = rs.align(rs.stream.color)
while(true):
    f = p.wait_for_frames()
    f = align.process(f).as_frameset()
    some something with f
```

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@jungseokhong Any other questions with the reply from dorodnic? Looking forward to your update. Thanks!

closed 2020-03-10 by dorodnic