realsenseai/librealsense #5909: T265 vslam with landmarks and relocalization

Dapeng-Zhang, 2020-02-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version | Win10| 
| Kernel Version (Linux Only)    |  ----| 
| Platform                                 | PC |
| SDK Version                            |  2.32.1                          | 
| Language                            |  C                           | 
| Segment			|  others                          | 

### T265 vslam with landmarks and relocalization

I'm extensively testing T265 and its v-slam.
The testing is done on a floor of 20m x 30m with several rooms. Height changes are ignored. T265 is attached to the top of a helmet, shooting towards ceiling during the test. An operator takes this helmet and walk around the floor.  
The example ar-advanced from "realsense SDK" is modified so that multiple static nodes can be added by key pressing during the test. 7 landmarks (rs2::set_static_node) are marked out with hopping that they will be used for re-localization and consequently long-term drift will be minimized. 
However, as shown in the test results, two rooms are drifted away in about 5 minutes walking, re-localization never happened during the test.
Could someone help to check if I have blind points (mistake) during this test?
Can T265 do real slam with map optimization at the loop points?
Many thanks

[image]

MartyG-RealSense, 2020-03-03:
I took another look at your case after our previous discussions at the Intel Support site's RealSense forum about this question.

T265 expert @radfordi said in another discussion that the T265 tries to relocalize a few times a second.  So relocalization should be occurring during your test.  The only reason I could think of why relocalization would not occur at all is if it had been disabled.

There is an option called **RS2_OPTION_ENABLE_RELOCALIZATION** which I believe is True by default.  The T265 documentation says about it:

"[Relocalization] allows the device to solve the Kidnapped Robot Problem, i.e. it will allow connecting the current map to a loaded map or connecting the current map to itself after accumulating a large drift, say after closing a large loop or covering the camera and walking around.  This is independent of jumping the pose.  **When fooled this feature can lead to much larger errors than are likely via the basic mapping**".

Dapeng-Zhang, 2020-03-04:
Many thanks @MartyG-RealSense. 
The new inputs are added into the main post.

MartyG-RealSense, 2020-03-04:
Thanks so much for your code contribution!  I will re-post the link to the zip below so that others reading this message can find it easily.

https://github.com/IntelRealSense/librealsense/files/4285419/rs-ar-advanced_static_nodes.cpp.zip

I believe that the triggers for a relocalization occurring can include:

- Travelling in a circle
- Covering up the camera
- Accumulating excessive drift
- Loss of power

Regarding the travelling in a circle condition, it looks as though the travel on your chart mostly involves travel in a straight line with an occasional turn to change direction.

Dapeng-Zhang, 2020-03-04:
@MartyG-RealSense Many thanks for your comments!

Actually I'm doing an evaluation of several products, to see if they can do full SLAM without long-term drift. I wonder if the following conclusion can be drawn, or is there a blind point (mistake) in my evaluation? 

T265 can provide quite accurate pose stream (or odometry). The documentation of T265 explains its functionality as a full slam. However, the experiments show that T265 can not reach a state that the mapping is 100% done and all movement afterwards on the known map has almost no accumulated (long-term) drift.

MartyG-RealSense, 2020-03-04:
I think a certain amount of drift after travel is inevitable, though there are things that can be done to improve results.  For example, drift may be reduced if the robot that the T265 is attached to initially moves faster than normal when it begins motion and then reduces to its normal speed.  This helps the T265 to understand its surroundings by giving it enough motion.

The link below is an excellent reference regarding drift.

https://github.com/IntelRealSense/librealsense/issues/3970

Dapeng-Zhang, 2020-03-05:
Thanks for your inputs @MartyG-RealSense 
I did the third test with confidence information. At 90% of trajectory come from the tracker confidence 3. Please reference the longConf.pdf in the main post.

MartyG-RealSense, 2020-03-05:
I had a look at the chart that you kindly provided.  I wonder if it would be possible to disable pose jumping and see if that improves your results further.

Dapeng-Zhang, 2020-03-09:
Thanks again for your inputs @MartyG-RealSense 
I did the 4th test with disabled pose jumping. 
The long-term drift is illustrated in the main post....

MartyG-RealSense, 2020-03-09:
Might there be an environmental factor that is contributing to your drift?  This could include reflective floors or excessive texture in the scene.  The 'Challenges' section of Intel's Visual SLAM white paper discusses these.

https://dev.intelrealsense.com/docs/intel-realsensetm-visual-slam-and-the-t265-tracking-camera#section-44-challenges

Dapeng-Zhang, 2020-03-09:
Here is a short video from one of the fisheye. 
[short.mp4.zip](https://github.com/IntelRealSense/librealsense/files/4306537/short.mp4.zip)

We carefully choose the T265 shooting direction.

MartyG-RealSense, 2020-03-09:
In the video it looks as though the T265 is pointing straight upwards and observing the ceiling as it moves, yes?

I wonder if the repeated exposure to bright glare from the ceiling lights as the camera lens passes directly under each one could be repeatedly throwing the results off.  The T265 would not be able to discern texture in the area of the light glare at these moments of strongest light concentration, as highlighted in the 'Challenges' section.

Dapeng-Zhang, 2020-03-10:
@MartyG-RealSense 
Yes, it is shooting upward, which is mentioned in the very first post at the second sentence two weeks ago. 

Shooting upward does not come easily. We systematically compared the results of 
           shooting forward,
           shooting backward,
           shooting sideways,
           and shooting upward. 
At the end, shooting at the ceiling generates the best results.

The lighting condition you mentioned will have a direct impact on the short-term drifts, which we never complained. The whole story is about long-term drift, landmarks, and loop detection and optimization.

MartyG-RealSense, 2020-03-10:
Thank you very much for your further details.  I have been advised that a dedicated T265 support engineer will be handling all T265 cases on the GitHub from now on, so I will have your case assigned to them.  Good luck!

Dapeng-Zhang, 2020-03-10:
Many thanks & Best Regards! @MartyG-RealSense

shsteven, 2020-04-24:
> Thank you very much for your further details. I have been advised that a dedicated T265 support engineer will be handling all T265 cases on the GitHub from now on, so I will have your case assigned to them. Good luck!

Is there any updates to this issue, or suggested workarounds?

SammyBravo, 2020-05-08:
As of yet I do not have a clear description of the difference between pose jumping and relocalization. For example, in t265.md it states:

"RS2_OPTION_ENABLE_POSE_JUMPING - This option allows the device to discontinuously jump its pose whenever it discovers its pose is inconsistent with one it has given before. For example, after walking in a circle or covering the camera (getting a small drift) and uncovering it. Currently this will **only affect the translation and not the rotation** nor any of the velocities or accelerations."

This is not mentioned for "RS2_OPTION_ENABLE_RELOCALIZATION", does this mean relocalization _also_ changes the rotation quaternion in addition to translation? 

What are the differences between pose jumping triggers and relocalization triggers?

Any information will be really helpful!!

neilyoung, 2020-05-08:
During a tracking session you can see these pose jumps very impressively if enabled: The position really jumps, and in fact it mostly jumps to exactly the place you would expect it to be.

RS2_OPTION_ENABLE_RELOCALIZATION is IMHO (just heuristic) necessary in case you have loaded a stored map and the device notices to have seen this scene again. Then the current mapping with be aligned w.r.t. the loaded map.  Also, if the algorithm detects loop closing it might change its mind regarding the environment. I think w/o this option you would not be able to retrieve static nodes.

I'm having hard coded both flags to true, but if someone would like to test the difference in behaviour, I would be ready to add these to the command line of my app:

https://neilyoung.serveblog.net/

I unfortunately don't currently have the time to do so.

SammyBravo, 2020-05-08:
Thanks @neilyoung I just had a look and this appears to be correct. Pose jumping is physically changing the translation, rotation is left unchanged. Relocalization, albeit harder to trigger, seems to be aligning the static nodes to the current reference frame and not the other way round. i.e. Re-localization **does not** change the pose at all and only effects the static nodes. 

What has me stumped then, is that even by turning off all the options ("Enable Mapping", "Pose Jumping", "Enable Relocalization", "Dynamic Calibration"), I cannot seem to make the T265 drift in rotation. eg. if I cover the sensor completely, giving a "low" confidence then moving it and uncovering the sensor it regains it's correct yaw rotation. I understand pitch is absolute relative to gravity, but yaw should eventually drift if there is no image/feature based relocalization. 

Any ideas about this?

neilyoung, 2020-05-08:
> Relocalization, albeit harder to trigger, seems to be aligning the static nodes to the current reference frame and not the other way round. i.e. Re-localization **does not** change the pose at all and only effects the static nodes.

Yes, this seems to be correct. The stored position is corrected w.r.t. the current reference frame, you don't have anything else :) That being said, a world coordinate remains the same.

 > if I cover the sensor completely, giving a "low" confidence then moving it and uncovering the sensor it regains it's correct yaw rotation. I understand pitch is absolute relative to gravity, but yaw should eventually drift if there is no image/feature based relocalization.
Any ideas about this?

IMHO the cam has an IMU, which contains of accelerometer and gyro. So the rotation should be very well seen by this.

> Do you think the rotation might be effected?

Not sure if a loop closing effects the rotation component, but relocation generally surely does. Think about this: You start your initial mapping at a given translation and rotation. You set a static marker at a certain point (stores translation and rotation). Then in another run with this map you start from a completely different point with a completely different rotation. However, if the algorithm relocates, since it recognizes the scene, , it gives you the pose of your stored static node, now in the reference frame of the new run. With surely different translation and rotation as stored. 

BTW: If you wanted to check my solution: My server is currently offline. I was forced to re-organize my network in order picture a DMZ. The blog was lost on the way :)

neilyoung, 2020-05-08:
PS: Online again

closed 2024-04-01 by MartyG-RealSense