realsenseai/librealsense #5911: Weird time reported by realsense

kevindehecker, 2020-02-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Ubuntu 18   | 
| Kernel Version (Linux Only)    |  5.3.0-40-generic  | 
| Platform                                 | PC  |
| SDK Version                            |  2.32.1                         | 
| Language                            | n.a.                        | 
| Segment			|  Robot                        | 

### Issue Description

The time is not going up like it should, also it does not start at 0. Video:
https://photos.app.goo.gl/juH3iArukHUxjJTR8

Background info:
Last few months I've been running firmware 5.11.06.250 on kernel 4.15.0-55 successfully, including metadata and hardware clock time that makes sense. After the fix of DKMS on kernel 5.x mentioned in #4586, I updated everything to see if I can get everything to work again. DKMS is installing fine now, but unfortunately the time reported in the realsense is said to be the "global time" (so not the hardware clock...?). Sometimes I also get error messages about the metadata, so it may be related to this, but at the moment those errors disappeared and I can't get them back just now.
I hope this may help to find the problem.

How can I fix this?

kevindehecker, 2020-02-25:
FYI: the settings in realsense viewer 
[image]

ev-mp, 2020-02-25:
@kevindehecker , since [v2.22](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features--improvements-8) a new inter-sensor/camera timestamp synchronization mechanism has been introduced - [Global Time Stamp](https://github.com/IntelRealSense/librealsense/pull/3909) that automatically applies to all D400 Depth cameras.

The feature is configurable separately per each sensor so you can control over how the active timestamps would be calculated by modifying [RS2_OPTION_GLOBAL_TIME_ENABLED](https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2220) option.
Note that when the Global TS is turned on you can still get the __original__ HW timestamp of the frame by using the metadata attribute query.

kevindehecker, 2020-02-25:
OK, sure, but did you look at the video? The time is standing still, sort of... That doesn't seem right at all.

ev-mp, 2020-02-25:
Please see the reply for #5912 - and the fix PR mentioned - https://github.com/IntelRealSense/librealsense/pull/5751.
Thus the timestamp report is a possible duplicate of #5319.

RealSenseCustomerSupport, 2020-03-03:
----------------------------------------------

Hi @kevindehecker,

Please install the latest build of RealSense SDK and let us know if you still experience the same issue/need any further assistance with it.

Thank you!

kevindehecker, 2020-03-03:
Just tried it, and the problem has indeed been solved. Awesome, tnx!

closed 2020-03-03 by kevindehecker