realsenseai/librealsense #5912: D435i Stereo images flicker mix

kevindehecker, 2020-02-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Ubuntu 18   | 
| Kernel Version (Linux Only)    |  5.3.0-40-generic  | 
| Platform                                 | PC  |
| SDK Version                            |  2.32.1                         | 
| Language                            | n.a.                        | 
| Segment			|  Robot                        | 

### Issue Description

Often when I start the rs viewer and show both stereo raw images, they get mixed up. Video: https://photos.app.goo.gl/3smGxRG7rhE56SbM8

It may be related to #5911, background info their applies here as well (i.e. this problem does not seem to happen on fw 5.11.06.250 on kernel 4.15.0-55 a.f.a.i.k )

ev-mp, 2020-02-25:
@kevindehecker , the flickering is weird indeed.
Can you record and provide a rosbag with a reproduction ?

As for the timestamp looping - a similar timestamp-related issue has been addressed in https://github.com/IntelRealSense/librealsense/pull/5751 available on the `development` branch (which was not part of v2.32.1, imho).
You can try building from source and provide feedback, alternatively this PR will be part of the next release.

kevindehecker, 2020-02-25:
Recording of the stereo flicker:
https://drive.google.com/file/d/1P_9CQEdactur7RAhhn0rvPYAFgq3zoOp/view?usp=sharing

I'll check the dev branch tomorrow, thanks!

kevindehecker, 2020-03-03:
Just tried to reproduce this stereo swap flicker problem with the newest releases (2.33.1 and FW 5.12.3.0 ), and it seems to problem has been solved. I've been eyeballing the images for a few minutes, and only twice I saw some flicker happen that could have been a swapped (or missed), not completely sure. In any case, with the previously the swap happened many times per second and that is definitely not the case anymore.

I'm leaving this open for a few more days, I'll keep it running in the background and see if something happens.

kevindehecker, 2020-03-13:
Seems to be fixed :)

closed 2020-03-13 by kevindehecker