realsenseai/librealsense #5913: Realsense sometimes not detected after reboot, permanently

kevindehecker, 2020-02-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Ubuntu 18   | 
| Kernel Version (Linux Only)    |  5.3.0-40-generic  | 
| Platform                                 | PC  |
| SDK Version                            |  2.32.1                         | 
| Language                            | n.a.                        | 
| Segment			|  Robot                        | 

### Issue Description

Sometimes if I reboot a computer with a realsense still connected with the USB cable that came delivered, the realsense is not detected anymore. Also not listed in lsusb. Additional reboots won't matter, I have to unplug and replug the realsense before it is seen again. This is a big problem for unattended systems installed in difficult places (which is an application of mine).

It was already there with fw 5.11.06.250 on kernel 4.15.0-55. I was hoping this problem would be solved with the latest updates and anything, but it has not.

What additional info can I acquire to debug this problem?

MartyG-RealSense, 2020-02-25:
The camera not being detected after a "soft reboot" (restarting the computer instead of doing a total shut-off with the power button and then pressing the power button to reboot) is a known issue on some computers.  The RealSense team did a lot of testing of the problem in previous cases.  It was thought that the problem may be related to the USB cable or USB port / controller.

Does the problem occur with every computer that you have tried it with, please?

If you are developing your own RealSense application, it is possible to incorporate an instruction called **hardware_reset()** into your scripting.  It resets the camera hardware in the same way that a USB unplug-replug does, without having to be present to do so.

The link below has a snippet of C++ code for performing a reset.  The code can be expanded upon to make reset conditional on certain events being met, such as when the number of invalid pixels reaches a certain level.

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

If you have an application with multiple cameras attached to the computer, it is also possible to reset by iterating through all attached cameras.

https://forums.intel.com/s/question/0D70P0000068r7PSAQ/d435-streams-work-once-then-crash?language=en_US

kevindehecker, 2020-02-25:
It happens on many Intel NUC7i3BNK, as well as on my Dell XPS 15. The cables are the one that came delivered.

I already have the hardware_reset going, but it doesn't work as it is not detected at all (so it cannot be reset this way). E.g. 
```
    rs2::device_list devices = ctx.query_devices();
    if (devices.size() == 0)
```
-> there are 0 devices.

MartyG-RealSense, 2020-02-25:
Another approach to a reset is to reset the entire USB port instead of the camera specifically.  This should work as it is not reliant on detecting the camera.  You can find out more by googling for 'linux usb reset script'.  I have seen a case with a RealSense user who successfully employed this method.

kevindehecker, 2020-02-25:
Tnx for the tip, gave that a try, but no luck. Again because the problem is that the device is completely not detected (i.e. listed in lsusb). So, I know from an identical system that the Realsense is supposed to be on `Bus 002 Device 003: ID 8086:0b3a Intel Corp.`, and when I try to reset that with `sudo ./usbreset  /dev/bus/usb/002/003` I get `Error opening output file: No such file or directory`.

FYI output of lsusb on problem NUC:

```
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 002: ID 1eaf:0004  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

FYI output of lsusb on working NUC:
```
lsusb
Bus 002 Device 003: ID 8086:0b3a Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 005: ID 1eaf:0004  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

MartyG-RealSense, 2020-02-25:
Considering your machines are NUCs, I wonder if it is an issue with the BIOS.  Before the RealSense team settled on USB as the likely explanation for detection failure on soft reboot, the common advice was to try a BIOS update.  More often than not, the PCs involved in those cases were NUCs, I recall.

There is a case in the link below with a very similar model to yours (BNK instead of BNH).

https://github.com/IntelRealSense/librealsense/issues/1615

MartyG-RealSense, 2020-02-25:
Going through the linked discussion again, it reminded me of the advice to check in your BIOS settings whether "xHCI" mode is set to Enabled.

https://github.com/IntelRealSense/librealsense/issues/1615#issuecomment-385585630

kevindehecker, 2020-02-25:
Thanks! I'll look into the bios and let you know if solved anything permanently, but I cannot change that easily with systems already in the field so it will take a while. However, that thread pointed me to `rtcwake -m off -s 30` which did the trick!

MartyG-RealSense, 2020-02-25:
Great news that **rtcwake** helped you!  I look forward to future test updates regarding the BIOS if you go ahead with trying that.   Good luck!

kevindehecker, 2020-02-26:
I tried al your suggestions regarding the NUC7i3BNK (which has the NUC7i3BNB main board) bios: 
- disabled legacy usb, 
- xHci setting is not available on NUC7, 
- tried the Portable Device Charging Mode -> Off,
- updated to latest BIOs version `BNKBL357.86.A.0081.2020.0114.1542`.

I still have the same problem that the realsense isn't listed in lsusb after a normal reboot. I have a NUC here on my desk that seems to have the problem almost always after every reboot, so that is nice for debugging I guess.

MartyG-RealSense, 2020-02-26:
My research indicates that RealSense cameras aren't the only USB device that can be non-detected on Linux after a soft reboot.  So expanding the investigation for fixes for soft-reboot detection problems in non-RealSense cameras may provide additional things to try.

For some people, the problem was fixed by updating the kernel.

In the non-RealSense cases I researched, a recurring theme was that the affected users had a USB hub.  Are your cameras attached to the NUC with a hub please?

kevindehecker, 2020-02-26:
In my cases the RS is directly connected to the NUC with the cable that came delivered with the RS.
I did try with some other USB devices attached to other ports, but the RS is the only one not coming back after a reboot.
On the NUC in question I'm running 4.15.0-88-generic, I'll update that to 5 and report back tomorrow.

kevindehecker, 2020-02-27:
Also with kernel 5.3.0-40-generic on the NUC7i3, the reboot problem persists. Any other suggestions? I mean I can work with the rtc reboot trick, but willing to help debug this further is possible.

MartyG-RealSense, 2020-02-27:
Given that **rtcwake** works for you, I wonder if something sleep-related is happening in Linux during boot.   Perhaps try setting all the power-saving options to 'Never'.

RealSenseCustomerSupport, 2020-03-02:
----------------------------------------------

We have a newer viewer 2.33.1 and FW just released with below fix

**FW 5.12.3.0 for D4xx Devices** Stability and Power Management fixes and enhancements

Please see if this can fix the problem you have now

MartyG-RealSense, 2020-03-02:
@RealSenseCustomerSupport   Thanks so much for the notification of the new release, firmware and stability fix!  I'll share that around.

As 5.12.3.0 is not on the firmware version download page at the time of writing this, I'm guessing that it should be installable automatically as the Recommended Firmware in 2.33.1's version of the RealSense Viewer's "More" option?

kevindehecker, 2020-03-02:
> We have a newer viewer 2.33.1 and FW just released with below fix

Updated... no luck again. The Realsense is still not showing up after every reboot.

> Perhaps try setting all the power-saving options to 'Never'.

There were not a whole lot of settings that were not already set to max performance etc. Played around a bit with some "S4/S5" state settings but it didn't change much.

Even more unfortunate; I'm also having some trouble with the rtc reboot work around, the NUCs seem to not always get back online after this command... These systems are still running original bios firmware, (difficult to update since I cannot easily reach these systems anymore, and bios update cannot be done from Linux a.f.a.i.k.), so that may cause the rtc problems. It means this issue just became a problem again :(

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

Hi kevindehecker,
Could you confirm the firmware is already updated to 5.12.3?
you can use viewer's info tab to check it

kevindehecker, 2020-03-04:
Confirmed:
[image]

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

Hi kevindehecker,
Please see if you have a USB3 hub with short cable length (as short as possible) e.g. 10 cm
only plug the D435i to it then plug the hub to PC USB3 port
use the latest FW 5.12.03 then reboot the PC to see if help

This did help on one case that one PC not cut the power during the reboot
That issue can be fix by running rtcwake that MartyG-RealSense already suggested
or add a USB3 hub. in this case, the Hub act like a buffer that allow the enumeration process can
move more reliability

kevindehecker, 2020-03-06:
I found some cheap USB3 hub with 3cm cable, and to my surprise this actually seemed to resolve the problem in the sense that the realsense device is now properly detected at least 20 reboots in a row! Progress :)

Haha, and a bit ironic, because previous advice was to keep hubs far away from realsenses. Indeed I get other weird problems now, like frame losses etc, when actually using the realsense. I'll arrange a better USB3 hub and retest.
(Needless to say that adding a hub to an already cramped device is not my preferred solution here anyway.)

In other news; I found the rtcwake solution has some problems because sometimes Ubuntu spends a long time on shutting down (like 10 minutes), and if before that the shutdown the rtc triggered already it doesn't do anything (and won't automatically retrigger later after shutdown). So after setting the rtc trigger to 30 minutes it seemed to solve that issue. (again, workable, but not ideal for obvious reasons)

RealSenseCustomerSupport, 2020-03-09:
----------------------------------------------

Hi kevindehecker,
Yes, basically we recommend only plug one camera to one USB3 port if you required high fps and resolution. but for multi-camera usage (need to lower the fps and resolution) we would recommend to use powered hub to provide sufficient power for multiple camera running simultaneously.
Your case is specially that not able to duplicated on the NUCs we had with Ubuntu 18 or 16  installed..
There could be many factors would cause this issue and adding a hub is kind of trick that can help in your case.

closed 2020-03-20 by RealSenseCustomerSupport

kevindehecker, 2020-03-20:
Hmm closed? I don't think the usb hub is a permanent solution here. I hope. Right?

connoranderson, 2020-03-25:
@RealSenseCustomerSupport is this actually resolved? @kevindehecker which USB hub did you use?

MartyG-RealSense, 2020-03-25:
@connoranderson   I'm not @kevindehecker but I thought you would like to know that when Intel tested RealSense cameras with a mains powered USB 3 hub for their white paper document on use of multiple cameras, they successfully used the **AmazonBasics** 4-port powered hub.

cowboygary, 2020-06-26:
@RealSenseCustomerSupport What's actually the solution?

mhudnell, 2021-08-31:
Unfortunately the `hardware_reset()` can't be performed in this case, sine as @kevindehecker described, the camera is not detected at all by the computer (not even listed by `lsusb`).

We're going to try removing our nightly reboot, leaving the application running all the time, and only restart when absolutely necessary, like when the kernel panics.

kevindehecker, 2021-08-31:
> Also note: The rtcwake command didn't fix it for me, I experienced the same problem as with the /sbin/reboot command.

Just checking: did you use an USB port that gets completely powered off at shutdown? Many ports nowadays keep power, for charging purposes I suppose, so you have to pick one that doesn't. Not sure what other hardware you are using, on our NUCs there's a small power symbol to indicate whether it stays powered on. I believe the usb charging behavior can even be selected through the BIOS. And of course an (externally) powered hub probably also doesn't work.

For your information, we enrolled the rtcwake reboot work around to our product (hundreds of units in the field) and it seems to work reliably.

MartyG-RealSense, 2021-09-01:
@kevindehecker  Thanks very much for your advice to @mhudnell   :)