realsenseai/librealsense #5914: Error when changing gain D435

kevindehecker, 2020-02-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Ubuntu 18   | 
| Kernel Version (Linux Only)    |  5.3.0-40-generic  | 
| Platform                                 | PC  |
| SDK Version                            |  2.32.1                         | 
| Language                            | n.a.                        | 
| Segment			|  Robot                        | 

### Issue Description
With the latest SDK I get an error when setting:
```
    auto gain_rng = rs_dev.get_option_range(RS2_OPTION_GAIN);
    rs_dev.set_option(RS2_OPTION_GAIN,gain_rng.min);
```

Error:
```
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  set_pu(id=4) failed! Last Error: Input/output error
```

The error happens in the realsense viewer as well:
[image]

ev-mp, 2020-02-25:
@kevindehecker hello,
Changing the (depth sensor)  gain control requires the Auto-Exposure to be set off by design.
Duplicate of #5753

kevindehecker, 2020-02-25:
Aha thanks for pointing that out. So that's changed then since the last time I updated. That requires some hefty changes on my side, because I used the fact that the auto exposure would not touch the gain. I assume the gain is now changed when AE is enabled?

ev-mp, 2020-02-25:
That's correct. The Depth AE is given an exclusive control over Exposure and Gain to adjust for light conditions.

RealSenseCustomerSupport, 2020-02-28:
----------------------------------------------

Hi @kevindehecker,

Do you need any further clarifications regarding this topic or it could be closed?

Thank you!

kevindehecker, 2020-02-29:
Nope that's OK. Thanks again!

closed 2020-02-29 by kevindehecker