realsenseai/librealsense #5917: Plans to support Bazel?

TheButlah, 2020-02-25:
# Feature Request
Maybe this is a pie in the sky feature, but would it be possible to support bazel? My team is looking to transition to it and we are hoping that our upstream dependencies provide the necessary `WORKSPACE` and `BUILD` files.

dorodnic, 2020-02-26:
Hi @TheButlah 
Thanks for the proposal. No plans at the moment, but we will evaluate it internally. Also, the project is open to community pull-requests

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

Hi @TheButlah, if you don't need further help. We will close this issue in 7 days.

closed 2020-03-05 by TheButlah

mawallace, 2020-09-21:
@TheButlah did you ever manage to get this working with Bazel?

TheButlah, 2020-09-21:
I managed to get it working by hard-coding the path to the apt-installed package. This is a bad hack, but it works as long as you don't need to worry about multiple distros and you are ok with a brittle dependency on apt (its version is not managed by bazel). I would greatly appreciate proper bazel support however - even just a separate repo giving an example of a build with bazel would be great!

mawallace, 2020-09-21:
Got it. 

I'm probably going to try to get an example working using https://github.com/bazelbuild/rules_foreign_cc to build librealsense with CMake and then pick it up into Bazel to be used as a dependency -- happy to share that if it works out, feel free to ping me if I forget.

I may also, at some point, try to get a proper Bazel BUILD working for this, in which case I'll happily share.

prantoran, 2020-11-24:
> Got it.
>
> I'm probably going to try to get an example working using https://github.com/bazelbuild/rules_foreign_cc to build librealsense with CMake and then pick it up into Bazel to be used as a [...]

Hi @mawallace, I am also trying the same approach, were you able to use RealSense inside Bazel?

Currently, I am looking into the http_archive rule for downloading the RealSense sdk and initiating the cmake build instructions.

prantoran, 2020-11-26:
I was able to include the shared objects and header files without CMake build in Bazel, but I guess it will not be portable. I followed the code design of [MediaPipe](https://github.com/google/mediapipe). Here are my steps:

WORKSPACE:
```
new_local_repository(
    name = "linux_realsense",
    build_file = "@//third_party:realsense_linux.BUILD",
    path = "/usr/local",
)
```
third_party/realsense_linux.BUILD:
```
licenses(["notice"])  # BSD license

exports_files(["LICENSE"])

cc_library(
    name = "realsense",
    srcs = glob(
        [
            "lib/librealsense2-gl.so",
            "lib/librealsense2-gl.so.2.40",
            "lib/librealsense2-gl.so.2.40.0",
            "lib/librealsense2.so",
            "lib/librealsense2.so.2.40",
            "lib/librealsense2.so.2.40.0",
            "lib/librealsense-file.a",
        ],
    ),
    hdrs = glob([
        "include/librealsense2/*.h*",
        "include/librealsense2/**/*.h*",
        "include/librealsense2-gl/**/*.h*",
    ]),
    includes = [
        "include/",
    ],
    linkstatic = 1,
    visibility = ["//visibility:public"],
)
```

third_party/BUILD:
```
alias(
    name = "realsense",
    actual = "@linux_realsense//:realsense",
)
```

Now, I added the dependency as a deps attribute in my projects's BUILD:
```
cc_library(
    name = "main_gpu",
    srcs = glob(["**/*"]),

    deps = [
        "//third_party:realsense",
	],
)

```

Then I could use RealSense by including in this way:
```
#include <librealsense2/rs.hpp> 
int main() {}
```