realsenseai/librealsense #5918: Application crashed with runtime error .RS2_USB_STATUS_OTHER

zainmehdi, 2020-02-26:
| Camera Model | T265 |
| Operating System & Version | Raspbian Buster |
| Platform | Raspberry Pi 4 |
| SDK Version | 2.32 |

### Issue Description
I have been having runtime error after the tracker starts. I couldnt find any information for why this is happening. Can anyone provide any information on what might be the cause.

Error : failed to submit request , error: RS2_USB_STATUS_OTHER

@MartyG-RealSense please provide some inisght if possible. Thanks

MartyG-RealSense, 2020-02-26:
The code preceding line 2100 in tm-device.cpp where the exception is thrown suggests to me that the error is triggered because the status of the USB is not currently set to 'SUCCESS' when submitting a request to the USB.  

Version 2.32.1 of the SDK did a radical rewrite of how the T265 is handled.  Would it be possible please to try version 2.31.0 and see if the problem disappears in that slightly older version?

zainmehdi, 2020-02-26:
Yes I can try with v2.31. Never had this issue with 2.31 before that.

zainmehdi, 2020-02-26:
Sadly I am unable to debug in Version 2.31. Strangely when I run 2.31 in debug mode it gives me unable to load file to device error.

MartyG-RealSense, 2020-02-26:
Yes, I see that you have mentioned about this problem before.

https://github.com/IntelRealSense/librealsense/issues/5725

closed 2021-03-17 by RealSenseSupport