realsenseai/librealsense #5919: Cannot get serial number of a frame in callback mode when playback

wangroger0801, 2020-02-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version | ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  Linux ubuntu 5.3.0-28-generic             | 
| Platform                                 | PC   |
| SDK Version                            |  development branch (7b4b3ef)                          | 
| Language                            |  C++  }                          | 
| Segment			|    multi-camera software  | 

### Issue Description
Hello,

We found our program cannot get the serial number from the frame when using the callback mode. It only occurred when the program were playing the bag file as a recording input.
More specifically, this line of code is hanging and blocking this callback event moving forward. 

```
auto camCallback = [&](const rs2::frame& frame){
        if (frame){
               ... ... //always execute
                auto serial = rs2::sensor_from_frame(frame)->get_info(RS2_CAMERA_INFO_SERIAL_NUMBER); //this line
              ... ... //never execute
```

Commenting out this line, our program works. But in this case we can only use one playback file input. Because we can not identify the owner of the frame. 

I was following the record-playback example for the replay part. 
Any clew how to fix this issue?
Thanks

MartyG-RealSense, 2020-03-17:
I apologise for the delay in receiving a response to this question.   Do you still need assistance please?   Thanks!

wangroger0801, 2020-03-17:
hello @MartyG-RealSense ,
Yes, we still needs help, thanks. Is it the proper way of playback from multiple recording files?
Currently I had a workaround for this issue, by writing serial numbers in hard code for each callback  definition which will leads to be not able to read recording files from other cameras. 
Thanks

MartyG-RealSense, 2020-03-17:
Callback programming is somewhat outside of my experience but I tried to research similar issues that have been reported in order to provide some guidance about the possible problem with your own script.

It sounds as though you are using multiple cameras, as suggested by the "multi-camera software" line in your specification at the top of the discussion and your use of serial numbers.  Is that correct please?

And are you using the **rs-multicam** example program?

wangroger0801, 2020-03-17:
Thanks for your reply, @MartyG-RealSense 
No, I am basically modifying **rs-callback** as the base of the frame piping method and then adding an optional input that reading from playback bag files. The reason is that we need a high frequency of IMU reading for estimating the orientation as**rs-motion** is following. The serial number are used for tracking the owner of a frame and then tag it.

Thanks.

MartyG-RealSense, 2020-03-17:
I carefully looked over your code and compared it to lines 40 to 60 of **rs-multicam**, which features pretty much the same line for getting the same line as you do.  I could not identify a definitive cause for your problem though.   But that is more a reflection of my level of programming knowledge.

I note that you call 'frame' in your serial number instruction.  Did you define frame earlier in the script, like **rs-multicam** does with the 'new-_frames_ variable that it uses later in its own serial number retrieval instruction?

https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp#L40

I will tag @dorodnic into the conversation in that hope that he can provide guidance for your programming, as he is the SDK Manager.

wangroger0801, 2020-03-17:
Using playback in callback mode is working. The issue is that the callback will only execute till the line that get the serial number from a frame. it will not execute the following lines. 
It actually not multi-camera related issue.  I put some parts of my code here shows how I start the stream. 
```
// ... in main function
std::map<std::string, rs2::frame>collectFrame;
auto camCallback = [&](const rs2::frame& frame){
        if (frame){
               ... ... //always execute
                auto serial = rs2::sensor_from_frame(frame)->get_info(RS2_CAMERA_INFO_SERIAL_NUMBER); //hang at this line
              ... ... //never execute
              collectFrame[serial] = frame; //for example: label the frame by its serial number. 
       }
}
std::string filePath = "./file.bag";
std::cerr<<"playback mode from: " << filePath << std::endl;
auto pipe = std::make_shared<rs2::pipeline>();
rs2::config cfg;
cfg.enable_all_streams();
cfg.enable_device_from_file(filePath);
rs2::pipeline_profile profile= pipe->start(cfg,camCallback);  

// do something

while(true){
//process the collected frames
}
```
I wonder whether it is related with **`std::make_shared`** . I tracked the calling sequence, in this function, it did  `make_shared`  to the sensor for another time. The first time is at creating the pipe object. I don't know how exactly those pointers are handled. 
```
    inline std::shared_ptr<sensor> sensor_from_frame(frame f)
    {
        std::shared_ptr<rs2_sensor> psens(f.get_sensor(), rs2_delete_sensor);
        return std::make_shared<sensor>(psens);
    }
```

Thanks again for your support.

MartyG-RealSense, 2020-03-17:
@wangroger0801   Thanks so much for sharing further details, as it all helps to diagnose problems.   I do not want to lead you down the wrong path by making guesses that are not backed by experience, so hopefully one of the other RealSense GitHub team members with more advanced programming knowledge than mine will be able to help you find where the problem is occurring.  

I wanted to make sure that you got assistance, since your case has been open for a few weeks now.

MartyG-RealSense, 2020-03-28:
This case will be closed after the next 7 days  if there are no further responses.  Thanks!

wangroger0801, 2020-03-28:
> This case will be closed after the next 7 days if there are no further responses. Thanks!

I am still waiting for the assistance thanks.

MartyG-RealSense, 2020-03-28:
I took another careful look at your question from the beginning.

In your callback routine, I notice that you open a routine with the { bracket but in the quoted code snippet in the original message you do not show a } bracket closing the routine.  Have you got such a close-bracket in your script please?

This is how **rs-multicam** handles the retrieval of the serial number from its multi-camera setup:

[image]

wangroger0801, 2020-03-31:
Hello @MartyG-RealSense ,
Thanks for your reply. The code I showed was for showing where the issue occurred. It is not caused by the programming logic.  I said without that line, the program can compile and run. 
`                auto serial = rs2::sensor_from_frame(frame)->get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
`
You can probably reproduce the issue by modifying rs-callback example by loading the playback file to the pipeline and then try to get the frame serial number in the callback function.

Thanks.

MartyG-RealSense, 2020-03-31:
Thanks for the further information.  As with your "Multi Camera Issue" discussion though, I do not have a programming environment set up at the moment to test your code, and it is a subject currently outside of my programming knowledge anyway.

The best way forward may be to re-post your original question as a new issue so it appears at the top of the forum and then close this one.  I do apologise.

MartyG-RealSense, 2020-04-12:
Closed due to no further comments.

closed 2020-04-12 by MartyG-RealSense