realsenseai/librealsense #5924: Ubuntu 18 apt package version 2.32 simlik issue

danielecazzari, 2020-02-26:
### Issue Description
I noticed than on latest apt package of realsense library 2.32 there is a simlink issue:
    librealsense2.so is linked to librealsense2.so.2.33 instead of librealsense2.so.2.32
This cause the following issue when compiling (for reference to other developers):
    No rule to make target '/usr/lib/x86_64-linux-gnu/librealsense2.so'

To workaround the til fix will be available:
   sudo rm librealsense2.so
   sudo ln -s librealsense2.so.2.32 librealsense2.so

I was able to repro the issue bot on Ubuntu 16 performing apt upgrade and on a VM Ubuntu 18 after a fresh install.

---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D4435                                                    | 
| Firmware Version                   | N/A                                                        | 
| Operating System & Version |  (Ubuntu 16/18)                                      | 
| Kernel Version (Linux Only)    |  N/A                                                      | 
| Platform                                 | IEITank / VM                                          |
| SDK Version                            |2.32                                                        | 
| Language                            |  C++                                                | 
| Segment			|  Other                       |

ev-mp, 2020-03-01:
@danielecazzari hello,
We've identified and resolved an issue with the repository that should be fixed by now.
Please remove the currently installation and get the authorized debian version with 
```sh
dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge
sudo apt update
sudo apt-get install librealsense2-dev ...
```

ev-mp, 2020-03-04:
@danielecazzari, do you need more assistance with the issue?

danielecazzari, 2020-03-04:
@ev-mp Thanks for the fix. 
I tested it today and works. 
I'm closing this issue.

closed 2020-03-04 by danielecazzari