realsenseai/librealsense #5925: Recorded ros bag doesn't play

imran514, 2020-02-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435I  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Android OS 10  |                                          | 
| Platform                                 | Android  |
| SDK Version                            |   LibRealSense Version 2.32.1  | 

### Issue Description
I have recorded a ros bag file using "recording" sample app that comes with librealsense sdk and when i try to play the same file using "playback" app it doesn't play.

i get below errors.

> 

> `2020-02-26 20:15:04.182 9923-9923/? E/Zygote: isWhitelistProcess - Process is Whitelisted
2020-02-26 20:15:04.183 9923-9923/? E/Zygote: accessInfo : 1
2020-02-26 20:15:04.190 9923-9923/? I/lsense.playbac: Late-enabling -Xcheck:jni
2020-02-26 20:15:04.201 9923-9923/? E/lsense.playbac: Unknown bits set in runtime_flags: 0x8000
2020-02-26 20:15:04.208 9923-9923/? D/ActivityThread: setConscryptValidator
2020-02-26 20:15:04.208 9923-9923/? D/ActivityThread: setConscryptValidator - put
2020-02-26 20:15:04.230 9923-9923/? I/lsense.playbac: The ClassLoaderContext is a special shared library.
2020-02-26 20:22:29.168 9923-9923/com.intel.realsense.playback W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@7cf72ca
2020-02-26 20:22:29.215 9923-9923/com.intel.realsense.playback I/MultiWindowDecorSupport: updateCaptionType >> DecorView@23b9122[], isFloating: false, isApplication: true, hasWindowDecorCaption: false, hasWindowControllerCallback: true
2020-02-26 20:22:29.216 9923-9923/com.intel.realsense.playback D/MultiWindowDecorSupport: setCaptionType = 0, DecorView = DecorView@23b9122[]
2020-02-26 20:22:29.227 9923-9923/com.intel.realsense.playback W/lsense.playbac: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (greylist, reflection, allowed)
2020-02-26 20:22:29.227 9923-9923/com.intel.realsense.playback W/lsense.playbac: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (greylist, reflection, allowed)
2020-02-26 20:22:31.545 9923-9923/com.intel.realsense.playback D/ViewRootImpl@3160a2b[MainActivity]: setView = DecorView@23b9122[MainActivity] TM=true MM=false
2020-02-26 20:22:31.551 9923-10516/com.intel.realsense.playback D/librs: Librealsense VERSION: 2.32.1
2020-02-26 20:22:31.552 9923-10516/com.intel.realsense.playback E/librs: Failed to create ros reader: Bag unindexed
2020-02-26 20:22:31.554 9923-10516/com.intel.realsense.playback I/chatty: uid=10298(com.intel.realsense.playback) identical 1 line
2020-02-26 20:22:31.556 9923-10516/com.intel.realsense.playback E/librs: Failed to create ros reader: Bag unindexed
2020-02-26 20:22:31.558 9923-10516/com.intel.realsense.playback E/librs playback example: streaming, error: Failed to resolve request. Request to enable_device_from_file("/storage/emulated/0/rs_bags/20200227_131037.bag") was invalid, Reason: Failed to create ros reader: Bag unindexed
2020-02-26 20:22:31.558 9923-9923/com.intel.realsense.playback D/SurfaceView: onWindowVisibilityChanged(0) true com.intel.realsense.librealsense.GLRsSurfaceView{f3a2021 V.E...... ......I. 0,0-0,0 #7f070036 app:id/glSurfaceView} of ViewRootImpl@3160a2b[MainActivity]
2020-02-26 20:22:31.566 9923-9923/com.intel.realsense.playback D/ViewRootImpl@3160a2b[MainActivity]: Relayout returned: old=[0,0][1080,2280] new=[0,0][1080,2280] req=(1080,2280)0 dur=6 res=0x7 s={true 502009720832} ch=true
2020-02-26 20:22:31.566 9923-10501/com.intel.realsense.playback D/OpenGLRenderer: createReliableSurface : 0x757ef071c0, 0x74e21c7000
2020-02-26 20:22:31.567 9923-10501/com.intel.realsense.playback I/AdrenoGLES: QUALCOMM build                   : def403d, I0e419467bc
    Build Date                       : 11/06/19
    OpenGL ES Shader Compiler Version: EV031.27.05.02
    Local Branch                     : 
    Remote Branch                    : refs/tags/AU_LINUX_ANDROID_LA.UM.8.1.R1.10.00.00.529.103
    Remote Branch                    : NONE
    Reconstruct Branch               : NOTHING
2020-02-26 20:22:31.567 9923-10501/com.intel.realsense.playback I/AdrenoGLES: Build Config                     : S P 8.0.11 AArch64
2020-02-26 20:22:31.568 9923-10501/com.intel.realsense.playback I/AdrenoGLES: PFP: 0x016ee187, ME: 0x00000000
2020-02-26 20:22:31.572 9923-9923/com.intel.realsense.playback D/SurfaceView: surfaceCreated 1 #8 com.intel.realsense.librealsense.GLRsSurfaceView{f3a2021 V.E...... ......ID 0,0-1080,2042 #7f070036 app:id/glSurfaceView}
2020-02-26 20:22:31.572 9923-9923/com.intel.realsense.playback D/SurfaceView: surfaceChanged (1080,2042) 1 #8 com.intel.realsense.librealsense.GLRsSurfaceView{f3a2021 V.E...... ......ID 0,0-1080,2042 #7f070036 app:id/glSurfaceView}
2020-02-26 20:22:31.575 9923-10501/com.intel.realsense.playback D/OpenGLRenderer: SurfaceChanged : 0x0 -> 0x0
2020-02-26 20:22:31.580 9923-10505/com.intel.realsense.playback W/Gralloc3: mapper 3.x is not supported
2020-02-26 20:22:31.582 9923-10501/com.intel.realsense.playback D/OpenGLRenderer: SurfaceChanged : 0x0 -> 0x74e80d4600
2020-02-26 20:22:31.589 9923-9923/com.intel.realsense.playback D/ViewRootImpl@3160a2b[MainActivity]: Relayout returned: old=[0,0][1080,2280] new=[0,0][1080,2280] req=(1080,2280)0 dur=3 res=0x3 s={true 502009720832} ch=false
2020-02-26 20:22:31.594 9923-9923/com.intel.realsense.playback D/ViewRootImpl@3160a2b[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
2020-02-26 20:22:31.594 9923-9923/com.intel.realsense.playback D/InputMethodManager: prepareNavigationBarInfo() DecorView@23b9122[MainActivity]
2020-02-26 20:22:31.594 9923-9923/com.intel.realsense.playback D/InputMethodManager: getNavigationBarColor() -855310
2020-02-26 20:22:31.594 9923-9923/com.intel.realsense.playback D/InputMethodManager: prepareNavigationBarInfo() DecorView@23b9122[MainActivity]
2020-02-26 20:22:31.595 9923-9923/com.intel.realsense.playback D/InputMethodManager: getNavigationBarColor() -855310
2020-02-26 20:22:31.595 9923-9923/com.intel.realsense.playback V/InputMethodManager: Starting input: tba=com.intel.realsense.playback ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
2020-02-26 20:22:31.595 9923-9923/com.intel.realsense.playback D/InputMethodManager: startInputInner - Id : 0
2020-02-26 20:22:31.595 9923-9923/com.intel.realsense.playback I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
2020-02-26 20:22:31.598 9923-9923/com.intel.realsense.playback D/ViewRootImpl@3160a2b[MainActivity]: MSG_RESIZED_REPORT: frame=[0,0][1080,2280] ci=[0,112][0,126] vi=[0,112][0,126] or=1
2020-02-26 20:22:31.598 9923-9923/com.intel.realsense.playback D/InputMethodManager: prepareNavigationBarInfo() DecorView@23b9122[MainActivity]
2020-02-26 20:22:31.598 9923-9923/com.intel.realsense.playback D/InputMethodManager: getNavigationBarColor() -855310
2020-02-26 20:22:31.598 9923-9923/com.intel.realsense.playback V/InputMethodManager: Starting input: tba=com.intel.realsense.playback ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
2020-02-26 20:22:31.598 9923-9923/com.intel.realsense.playback D/InputMethodManager: startInputInner - Id : 0
`

MartyG-RealSense, 2020-02-26:
The 'bag unindexed' error messages suggest that the bag recording process was not properly closed when it was ended.  If you download a sample bag file from the link below and try to play it back, that will help to determine whether the problem is with your recorded bag.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

imran514, 2020-02-26:
@MartyG-RealSense 

I am able to play most of those sample, all those files are less than 1GB, but the file i recorded is more than 3GB, does it causing the issue ?

How can i am make sure that the recording process is properly closed in the "recording" app ?

MartyG-RealSense, 2020-02-26:
If you can record a very short clip that is seconds in duration so that it is less than 1 GB in size, that should answer the question of whether size is an influence in the problem.  It is likely related to the file closing process though.

@dorodnic offers some advice about closing here:

https://github.com/IntelRealSense/librealsense/issues/3029#issuecomment-453114359

imran514, 2020-02-27:
@MartyG-RealSense 

Most of those comments from @drodnic are for cpp library.
Is there any way to make sure that the recording is properly closed in the android app ?

MartyG-RealSense, 2020-02-27:
I will link the authors of the recording example into this conversation to see if they can provide input.  @matkatz @alowenst01 ...  And @ev-mp might be able to help too.

imran514, 2020-02-28:
@matkatz @alowenst01 and @ev-mp 

Any help would be appreciated

inesdfelix, 2020-02-28:
I'm also having the same problem! When I record using the example "recording" from the android wrapper and then try to play back the file using Python returns the following error: "RuntimeError: Failed to resolve request. Request to enable_device_from_file("20200228_105609.bag") was invalid, Reason: Failed to create ros reader: Bag unindexed". 
Would really appreciate the solution to this! Thank you in advance

Murded, 2020-03-04:
@inesdfelix @imran514 Not a solution as to why you're bags are being unindexed.

However you can 'fix' unindexed bags using the rosbag reindex take a look [here](http://wiki.ros.org/rosbag/Commandline#reindex)

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

@inesdfelix @imran514 Did you get it through with the suggestions from Murded and MartG-RealSense? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-17:
----------------------------------------------

@inesdfelix @imran514 Any update from your side? Thanks!

inesdfelix, 2020-03-19:
It worked for me. Thank you!

MartyG-RealSense, 2020-03-19:
@inesdfelix   Great news that you were able to fix your problem.   Thanks for letting us know!

RealSenseCustomerSupport, 2020-03-20:
----------------------------------------------

@inesdfelix Glad to hear the good news.
@imran514 Did you get through also? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-27:
----------------------------------------------

@imran514 Will close this if no other questions. Thanks!

closed 2020-04-08 by RealSenseCustomerSupport