realsenseai/librealsense #5927: Different measurements, on different PCs, same S.O.

ajpernalete, 2020-02-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D35I  | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Win (10)   |                                         | 
| Platform                                 | PC  |
| VIEWER Version                            |  2.31.0                          | 
| Language                            |  python                          | 
| Segment			|  {Visión Artificial Scan }                          | 

### Issue Description

Hello, I have an error in the measurements of Realsense D435i, where in a Notebook I have measurements very close to reality at distances greater than 4M, but in an industrial computer I do not want to go beyond 4M. I created .JSON file to save the settings and nothing. The industrial PC has a USB 3.0 port and activates both depth image and color image good. Try increase the Laser signal and does not affect the measurement at all. I am testing directly on the RS-Viewer.exe Version 2.31.0, update the Firmware of the cameras. What could be other causes for this difference between measurements? Thank you.

Preset.JSON

{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "-0",
    "aux-param-colorcorrection11": "-0",
    "aux-param-colorcorrection12": "-0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "-0",
    "aux-param-colorcorrection6": "-0",
    "aux-param-colorcorrection7": "-0",
    "aux-param-colorcorrection8": "-0",
    "aux-param-colorcorrection9": "-0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-amplitude-factor": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280"
}

MartyG-RealSense, 2020-03-03:
I apologise for the delay in responding.

The effect of changing the Laser Power is that it makes the semi-random dot pattern projected onto a scene by the camera's IR Projector component (a separate component from the IR sensors) more visible if you increase laser power and less visible if you decrease laser power.  

When the dot pattern is projected onto flat surfaces with low or no texture detail such as doors, walls and desks, it can help the camera to analyse the depth of those objects.  So you might only get an improvement in measuring results by changing Laser Power if you were measuring these kinds of low-textured objects.

From your description of the problem though, it sounds as though there is not an error in the **values** of the measurements, and what you want to do is prevent the camera from creating data for distances further than 4 meters from the camera.

If you are using the RealSense Viewer for your measuring on the industrial computer, then by default the camera will try to see as far as it is able to (the 400 Series has a default depth sensing range of 10 meters).  The Viewer has a post-processing option though called the **Threshold Filter** which is disabled by default.

[image]

If you expand open the settings of the Threshold Filter and set the 'Max Distance' option to '4' and enable the filter, depth data that is more than 4 meters from the camera will be removed from the background of the image.  

[image]

ajpernalete, 2020-03-04:
Hi @MartyG-RealSense 

Thank you very much for the reply. Just as I solved the problem, it was a depth calibration. I solved it using the RS viewer.exe to Calibrate the depth. This week I'll go to install it again.

Regards.

closed 2020-03-04 by ajpernalete

MartyG-RealSense, 2020-03-04:
Thanks so much for providing an update.  Good luck with your work!