realsenseai/librealsense #5930: Native Realsense for Android CMake Issue

pdrey919, 2020-02-27:
Hi There, 
I just got the T265 tracking camera and have set up a desktop app and tested it, but I'd like to get it working on an android program I have. I understand it's not fully supported, but if I can just get the position data from it, that's plenty. 
I followed all instructions on editing the gradle file, and now I want to use the JNI as described [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/native_example) so I can write the methods in c++ and call them in my java program.
I don't have any CMakeLists.txt files though, so I can't follow the step of editing my app's Cmake files (it's all in java). Should I run CMake and configure/generate the sdk into my current project? Or will that compromise what I already have there in java?
I'm new to the whole concept of integrating C++ and Java, so forgive me if I'm misunderstanding the whole issue. Appreciate any advice you have about getting android working and thanks in advance.

closed 2020-03-01 by pdrey919

smartynenko, 2020-03-01:
did you get it to work? if so, what was the solution? thanks

pdrey919, 2020-03-01:
No, I built an app using the JNI but couldn't get the camera to work with android. I might need to write a usb driver for it, which I don't know how to do. 
Are you trying to do the same thing?

smartynenko, 2020-03-01:
I was able to make little progress and rebuilding library with tm2 code (#5681), so system can detect camera, but then it fails to boot it. At this time I am stuck on trying to write firmware image to the camera which fails to work in both native and java code

pdrey919, 2020-03-02:
I see. I didn't see there was another thread about this- surely someone in the community has gotten it to work. I hope you can too, and let me know if you get it working. I'll be trying more as well.