realsenseai/librealsense #5931: D435 firefly RK3399  USB3.0  NO Frames Received

azuredsky, 2020-02-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | {  D435 } | 
| Firmware Version                    | D4XX_FW_Image-5.12.2.100.bin | 
| Operating System & Version   |   { aarch64 GNU/Linux (Ubuntu 16.04)    | 
 Kernel Version (Linux Only)      | Linux firefly 4.4.77                               | 
| Platform                                  | firefly RK3399 |
| SDK Version                            |  {2.32.1 }                          | 
| Language                                |  {C++ }                          | 
| Segment			          |  {others }                          | 

### Issue Description
D435 RK3399  USB3.0  NO Frames Received,but USB2.0 can get frames
terminal:
 27/02 09:53:39,773 WARNING [547977838592] (backend-v4l2.cpp:1208) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!

[image]
when use USB2.0:
[image]

ev-mp, 2020-02-27:
@azuredsky , did you try to use self-powered USB hub?

azuredsky, 2020-02-28:
@ev-mp 

3Q, I will try it.

ev-mp, 2020-03-03:
@azuredsky , according to the original screenshot the host receives only partial frames (see the log window at the bottom of the image).
So once you've managed to rule out power-related issues the issue is isolated to host processing capacity.

There are certain things that you can check:
- CPU and Memory Utilization:
    - Make sure that the CPU is less than 100%. 
    - Check that you have a swap file defined ( at least 1Gb)
    - Make sure the SDK is built in release
- Realsense-Viewer uses a lot of OpenGl stuff:
    - Check whether less resource-hungry demos, such as `rs-depth` `rs-capture` can stream the data with USB3 connection
   - Experiment with OpenGl settings in viewer (upper right corner) to offload processing/rendering stuff to GPU

In case you're building with `-DRSUSB_BACKEND=true` - try to turn it off and rebuild. This will select a different (v4l) kernel driver to experiment with.

azuredsky, 2020-03-10:
@ev-mp ，I have solved it，by rebuild with cmake ../ -DBUILD_EXAMPLES=true  -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=Release，
add  -DFORCE_LIBUVC=true is must

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

@azuredsky Thanks for the feedback and glad to hear the issue resolved. Close this accordingly.

closed 2020-03-10 by RealSenseCustomerSupport