realsenseai/librealsense #5933: Android - CameraInfo.USB_TYPE_DESCRIPTOR

Murded, 2020-02-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Android | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  3.32.1  | 
| Language                            | Java  | 

### Issue Description
When accessing the CameraInfo.USB_TYPE_DESCRIPTOR the value is returned as 2.1 when using a USB C cable, is this expected behaviour?

### Description of what I'm trying to accomplish
I'm trying to set the frame rate of the streams depending on the type of connection the device is using, this is because I've noticed that there is freezing when the frame rate is too high on older devies (the ones that use the older connection (not type c)).

Any suggestions would be highly appreciated

RealSenseCustomerSupport, 2020-02-28:
----------------------------------------------

@Murded Do you mean you're using Type C USB cable? Please note that some kinds of Type C USB cable is not USB3 cable but USB2. Could you please double check the cable is USB3 or USB2?  Also please check the USB port on your android device is USB2 or USB3. Only when USB cable and USB port are both USB3, then you can get USB3 connection. Thanks!

Murded, 2020-02-28:
@RealSenseCustomerSupport Hi, thanks for such a quick response.

I have confirmed that the connection on the phone is USB 3 (See specification [here](https://www.gsmarena.com/honor_view_20-9468.php)) and I've also tried a variety of cables and I beleive that they are also USB 3 (I have connected the realsense camera to the desktop and there is no USB 2 listed in the realsense viewer)

The issue I'm seeing is that on my honor view 20 (Has a USB type c port) when I connect the camera (Using a C to C cable or [this adapter](https://www.amazon.co.uk/Ailun-Adapter-MacBook-ChromeBook-Devices/dp/B01MYPX7Y8/ref=sr_1_15?keywords=usb+type+c+otg&qid=1582891499&sr=8-15)) and view the infromation inside the realsense camera app (Obtainable from the Android app store) it is identified as USB 2.1, however if I run the depth stream at 640x480 30fps and colour stream at 640x480 30 fps there is no issue.

But when I connect the camera to a variety of older devices that use micro usb using [this adapter](https://www.amazon.co.uk/Rankie-R-1170-Female-Adapter-Convertor-Black/dp/B00YOX4JU6/ref=pd_sim_23_5/259-5522017-9989735?_encoding=UTF8&pd_rd_i=B00YOX4JU6&pd_rd_r=fa102a5c-fb0f-414c-b16d-558c25e18796&pd_rd_w=4N9gn&pd_rd_wg=F8s6n&pf_rd_p=bbc136fb-d98b-4822-a5fd-b92d15766975&pf_rd_r=7GSFYQGV8XV8A86Z71VW&psc=1&refRID=7GSFYQGV8XV8A86Z71VW) it is also identied as USB 2.1 however running both the depth and colour stream at 640x480 30 fps causes freezing.

RealSenseCustomerSupport, 2020-03-11:
----------------------------------------------

@Murded If the cable is recognized as USB3 cable, then "identified as USB2.1" issue on your honor view 20 might be caused by the phone port itself.  For your issue on other older devices, could you please provide the logcat and dmesg logs for the freezing issue? Thanks!

RealSenseCustomerSupport, 2020-03-17:
----------------------------------------------

@Murded Did you get the log for the issue on other older devices? Looking forward to your update. Thanks!

Murded, 2020-03-17:
> @Murded Did you get the log for the issue on other older devices? Looking forward to your update. Thanks!

Unfortunately as I'm currently working from home I don't have access to the device and cable in question to grab the information you requested

RealSenseCustomerSupport, 2020-03-27:
----------------------------------------------

@Murded Thanks for update! Please share with us the update when you get chance to access the camera. Thanks!

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@Murded Did you get chance to access the camera and any update? Thanks!

Murded, 2020-04-08:
> @Murded Did you get chance to access the camera and any update? Thanks!

Hi, Sorry I still don't have access to the device and I'm still not sure when I will.

You can close this issue if you like and I can re-open or reference this issue if needed in future

RealSenseCustomerSupport, 2020-04-09:
----------------------------------------------

@Murded Thanks for the reply! Will close it at this point. Yes, please feel free to create another one if you still have issues or questions later. Thanks!

closed 2020-04-09 by RealSenseCustomerSupport