realsenseai/librealsense #5935: Usin

ArturSchuetz, 2020-02-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Ubuntu 18.04.4 LTS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)  | 
| Platform                                 | PC  |
| SDK Version                            |  2.32.1  | 
| Language                            |  C++  | 
| Segment			|  others  | 

### Issue Description

I have two code examples added: The first one works perfectly fine while the second does not. But in order to use multiple cameras and a T265, I need to use the second approach. It worked for a previous SDK version but after an update it broke.

This is the error I get for the second example: rs2_pipeline_start_with_config_and_callback_cpp(pipe:0x558422e41460, config:0x558422e39b60, callback:0x558422e26de0):
    No device connected

adding this line returns the same error:
cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));

```
#include <librealsense2/rs.hpp>

#include <iostream>
#include <thread>
#include <mutex>

std::mutex g_data_mutex;

int main(int /*argc*/, char* /*argv*/[])
{
    rs2::context context;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe(context);

    // Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;

    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);

    try
    {
        auto callback_Tracking = [&](const rs2::frame& frame)
        {
            std::lock_guard<std::mutex> lock(g_data_mutex);

            if (auto fp = frame.as<rs2::pose_frame>()) {
                auto pose_data = fp.get_pose_data();
                std::cout << pose_data.rotation.x << ", " << pose_data.rotation.y << ", " << pose_data.rotation.z << ", " << pose_data.rotation.w << std::endl;
            }
        };
        pipe.start(cfg, callback_Tracking);
    }
    catch (const rs2::error & e)
    {
        std::cerr << "RealSense error calling: " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    }
    catch (const std::exception& e)
    {
        std::cerr << e.what() << std::endl;
    }

    std::this_thread::sleep_for(std::chrono::milliseconds(1000));
    return 0;
}
```

```
#include <librealsense2/rs.hpp>

#include <iostream>
#include <thread>
#include <mutex>

std::mutex g_data_mutex;

int main(int /*argc*/, char* /*argv*/[])
{
    rs2::context context;

    for(auto&& dev: context.query_devices())
    {
        std::string TrackingCameraName = "Intel RealSense T";
        if (std::string(std::string(dev.get_info(RS2_CAMERA_INFO_NAME)).substr(0, TrackingCameraName.length())).compare(TrackingCameraName) == 0)
        {
            rs2::config cfg;

            // Declare RealSense pipeline, encapsulating the actual device and sensors
            rs2::pipeline pipe(context);

            cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);

            try
            {
                auto callback_Tracking = [&](const rs2::frame& frame)
                {
                    std::lock_guard<std::mutex> lock(g_data_mutex);

                    if (auto fp = frame.as<rs2::pose_frame>()) {
                        auto pose_data = fp.get_pose_data();
                        std::cout << pose_data.rotation.x << ", " << pose_data.rotation.y << ", " << pose_data.rotation.z << ", " << pose_data.rotation.w << std::endl;
                    }
                };
                pipe.start(cfg, callback_Tracking);
            }
            catch (const rs2::error & e)
            {
                std::cerr << "RealSense error calling: " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
            }
            catch (const std::exception& e)
            {
                std::cerr << e.what() << std::endl;
            }
        }
    }

    std::this_thread::sleep_for(std::chrono::milliseconds(1000));
    return 0;
}
```

korzen, 2020-02-28:
I can confirm the issue, neither rs-multicam nor rs-tracking-and-depth examples from master branch work. The cameras work fine together in the viewer so its most likely the never ending story of the t265 initialization along with d4x5 camera...

<pre>RealSense error calling rs2_pipeline_start_with_config(pipe:0x5608b46230e0, config:0x5608b4470ac0):
    No device connected
</pre>

[2020-02-28-08_31_24.log](https://github.com/IntelRealSense/librealsense/files/4265771/2020-02-28-08_31_24.log)

stino78, 2020-03-08:
I have the exact same problem in Python, using the last lib realsense 2.33.1.1370

Can you help us with the problem?

andreaszwengINS, 2020-03-10:
I have the exact same problem.

T265, D415
Linux Mint 19.3
Kernel 5.3

leochien1110, 2020-03-11:
I have the same issue with D435 and T265, Ubuntu18.04, v2.33.1

andreaszwengINS, 2020-03-12:
I could fix the problem for me. I had `enable_device` and `enable_stream` in my code for every device. I removed `enable_device` and it works better than ever before. Maybe this method never worked correctly? Thats why rs-multicam also does not work anymore. We use one t265 in combination with one D415. So, by setting `enable_stream` it seems to automatically know which camera belongs to which pipeline. Not a proper workaround, but this is the reason, it stopped working for us.

RealSenseSupport, 2020-09-11:
Thank you for highlighting this issue while using T265 with other RealSense products. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue with the T265 other than the already completed PRs.

closed 2020-09-17 by RealSenseSupport