realsenseai/librealsense #5936: XCODE Swift C++ wrappers

omarfakhr, 2020-02-27:
| Camera Model                       | {D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |    MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | MacOs  |
| SDK Version                            |  {  }                          | 
| Language                            |  {Swift/C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description

Hi There

I managed to import library into XCODE project and wrap the function for pipeline.start() to use it in swift. here is how I did it:

in a CPP-Wrapper.h
```
@interface CPP_Wrapper : NSObject
    - (NSObject *)pipeline;
@end

```
in CPP-Wrapper.mm

```
@implementation CPP_Wrapper 
    - (NSObject*)pipeline {
        rs2::pipeline p;
        p.start();
}
@end
```

I call this from the AppDelegate: 
`let p = CPP_Wrapper().pipeline()`
And Voila: the camera starts. However, if I add the function :

`rs2::frameset frames = p.wait_for_frames();`
        
the timeout error occurs. I think it is not getting any data. If I do the program using CMAKE and only C++ the camera works, so I think there is nothing wrong with the camera, just my code.

Can you please help?

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

Have you enable the "USB" capability in "App Sandbox" section" in the page"Signing & Capabilities"?

https://developer.apple.com/documentation/xcode/adding_capabilities_to_your_app
https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_security_device_usb

omarfakhr, 2020-03-10:
Yes, I think I did. I also checked the boxes for Camera. I still get the same error.

RealSenseCustomerSupport, 2020-03-11:
----------------------------------------------

Please double check your code.
I have try call librealsense c API in swift code. it works.

Please find more error detail. for example, you have found the camera before pipeline::start
and please make sure the lifetime of objects in your snippet.

omarfakhr, 2020-03-11:
Would you be so kind as to send me an example? It would help a lot.

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

There is a simple sample code.
https://github.com/mengyui/librealsense/tree/swift/wrappers/swift
It may help you.

omarfakhr, 2020-03-20:
@RealSenseCustomerSupport Thanks a lot. That is great. However there are files missing. E.g. Preview Content Assets, and the *.a linker flags are also not available. How Do I create the .a files?

Another thing: Because IU do not have .a files, i embedded a dylib library, that I have been using in my previous project. Its version is 2.32.1. When I run the app with this library I get: 

>API version mismatch: librealsense.so was compiled with API version 2.32.1 but the application >was compiled with 2.33.1! Make sure correct version of the library is installed (make install)

thanks for the help

RealSenseCustomerSupport, 2020-03-20:
----------------------------------------------

If you try to use static library, you can disable `BUILD_SHARED_LIBS` in cmake phase, when build librealsense2 library.

If you want to use dynamic library, Please try follow this step. I have not test that.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md#packaging-your-application

omarfakhr, 2020-03-23:
@RealSenseCustomerSupport  Thanks a lot. That was really helpful.

closed 2020-03-23 by omarfakhr