realsenseai/librealsense #5938: Optimum SetUp of Filters in RS Viewer for Indoor Capture

DavidCadtec, 2020-02-27:
Using a D415.
Running on Ubuntu 16.04

### Issue Description
@RealSenseCustomerSupport
I'm struggling here trying to achieve an optimum configuration recording people walking indoor. In the range of 1 - 8m.
Basically i'm trying to setup the posto processing filters to record avoiding artifacts in the depth colorized stream.
Is there any tip to achieve the best result possible ?  
Should i turn on any (which ?) filters like threshold, space, temporal ...  or others ?
Tried to use some of the default that come already with RS Viewer... but for poor results...
Thanks !

MartyG-RealSense, 2020-02-28:
Though the D435 model is best suited to capturing motion thanks to its fast Global Shutter (compared to the slower Rolling Shutter on the D415), the D415 should still be able to handle human walking pace.  

Intel's tuning guide for the 400 Series cameras advises that if the scene has bright light then you may be able to reduce motion artifacts by setting exposure manually to near to 1 ms.

DavidCadtec, 2020-02-29:
@MartyG-RealSense 
Thanks for the advise ... fot the time being i will continue capturing the pedestrian materials for my research using my D415 but i'm considering buying a D435 or other (Zed ?) Depth Camera for some further tests ... is there any recent comparison Matrix between Depth Cameras features that you would recommend as a reference ? 

Also thank you for sending the link for the tuning guide ... really very useful ! Interesting reading with several tips !

MartyG-RealSense, 2020-02-29:
Intel have a 'Which camera is right for you' guide that compares the models, including the forthcoming new L515 lidar camera.  Near the bottom of the page is a feature comparison chart.

https://www.intelrealsense.com/which-device-is-right-for-you/

DavidCadtec, 2020-03-03:
@MartyG-RealSense 
Thanks again for you comprehensive reply and for sending those extra partners links ...
L515 really seems that it's going to be very appropriate in my area of research (pedestrian tracking indoors...) right ?
Just one more related question: will this be back compatible (in terms of SDK / wrapper programing ...) with codes already developed for 415 or 435 ?

MartyG-RealSense, 2020-03-03:
You are very welcome.  :)

L515 is best suited to use indoors or in controlled lighting conditions.  It is not vulnerable to motion blur and does not require recalibration during its lifetime.  So it does seem to be an appropriate choice for your application.

You may be interested in a recently released video demonstrating the new Cubemos Skeleton Tracking SDK for RealSense being used to track a mass crowd.  Cubemos is compatible with both the 400 Series cameras and L515.

https://youtube.com/watch?v=PENY6Q24c6A

Cubemos can be tried now with a 30 day free trial version.  A full licence is $75 USD in the official RealSense store.

**Video**
https://youtube.com/watch?v=PENY6Q24c6A

**Information page with trial download button**
https://www.intelrealsense.com/skeleton-tracking/

**Store link**
https://store.intelrealsense.com/buy-skeletal-tracking-sdk.html

In regard to L515's compatibility with the RealSense SDK: the RealSense SDK Manager has said that 99% of the SDK's functions will be supported for L515 out of the box at launch.  This is possible because the RealSense developers spent months adding support before the camera was even officially announced.

A set of new Visual Preset camera configurations for the L515 were also added in the latest 2.33.1 version of the RealSense SDK.

Existing programs for the 400 Series should be simple to convert.  Intel has said that the Cubemos software only took half an hour to get working on L515.

DavidCadtec, 2020-03-05:
Thanks @MartyG-RealSense  !
This Cubemos SDK is in definitely impressive !! 
In fact, what i need in my research is a subset of what this SDK seems to be able to provide ! I will experience the trial to check the capabilities... for sure.

Also, thank you for the explanations about the back compatibility of L515 for using code created for 400 series ! Your guidance is much appreciated Marty !

closed 2020-03-05 by DavidCadtec

MartyG-RealSense, 2020-03-05:
Hi @DavidCadtec I just wanted to add that the official RealSense blog has published a new article on skeleton tracking that contains a section speaking of the ease of converting 400 Series programs to L515.

https://www.intelrealsense.com/skeletal-tracking/

DavidCadtec, 2020-03-05:
Super. Thanks @MartyG-RealSense  !!

Basuki74, 2022-05-25:
Dear @MartyG-RealSense,

Since 2019 I have been developing my research system based on Cubemos Skeleton Tracker, and I just found out that currently Intel Realsense Camera is not supported by Cubemos SDK. How can I continue to use the Cubemos SDK?
I didn't have much time to start my research activities from scratch.
Thank you for your advise.

MartyG-RealSense, 2022-05-25:
Hi @Basuki74  The Cubemos Skeleton Tracking SDK was a product of the Cubemos company and was withdrawn from availability, so there is no option to continue with an existing licence or download trial versions, unfortunately.  The best option for a quick transition for your research project is likely to be to switch it to the similar RealSense-compatible **Nuitrack SDK** skeleton tracking software if it is possible for you to do so.

https://nuitrack.com/

Basuki74, 2022-05-25:
Dear MartyG,

Thank you very much for your quick response. Actually I've tried using
Nuitrack, but after I tried to implement it, the accuracy is less than
Cubemos. Skeletal joints generated by SDK are unstable for some distance
and also when the hands cover the body. Others, hand gesture cannot be
detected if the distance between humans and the camera is more than 1.5 m.
Those are many reasons why I want to continue using Cubemos.
I have also tried using MediaPipe, but as you know they can't support for
depth data.

I think Cubemos is very powerful for a skeletal joint tracing library.

 Is there any problem in the future if I still continue for using Cubemos?

Best regards,

Dwiki

MartyG-RealSense, 2022-05-25:
My understanding is that once the time limit on a Cubemos SDK trial version expires then it is no longer possible to purchase a licence key.  The installer file for the Cubemos SDK program itself still has a working download for Windows and Linux though at the link below.

https://dev.intelrealsense.com/docs/skeleton-tracking-sdk-installation-guide

Pages 23-24 of the Cubemos manual have information about license keys.

https://download-skeleton-tracking-sdk.s3.eu-central-1.amazonaws.com/GettingStartedGuide_2.3.2.pdf

Basuki74, 2022-05-25:
Thank you @MartyG-RealSense for your explanation.