realsenseai/librealsense #5940: Announcement:  Cubemos 'Skeleton Tracking SDK' clip (UPDATE: Free Trial Version)

MartyG-RealSense, 2020-02-28:
Hi everyone,

The official RealSense YouTube channel has added a clip of the new Cubemos 'Skeleton Tracking SDK', which is available from the official online RealSense store for a licence fee of $75 USD.

**UPDATE:  A free trial version is now available for download from the information page at the link below.**

https://www.intelrealsense.com/skeleton-tracking

***********************

**YouTube clip:**
https://www.youtube.com/watch?v=PENY6Q24c6A

**Store listing**
https://store.intelrealsense.com/buy-skeletal-tracking-sdk.html

Below is an earlier YouTube clip of the Cubemos skeleton tracking running on the forthcoming new RealSense L515 lidar camera. This clip debuted during CES 2020 as an example of the ease of converting programs for the 400 Series cameras to L515.

https://www.youtube.com/watch?v=PTMo4Dvya1Y

korzen, 2020-02-28:
Looks promising but it does not work with GPU (GTX 1080 driver 441.41, CUDA v10.1 installed) and the inference performance on the CPU (i7-4790K @4.4GHz) is rather poor. 

[image]

MartyG-RealSense, 2020-02-28:
Thanks very much for your feedback about the problem that you had.  It looks as though you are using the Cubemos SDK's 'charp-interactive' sample program in the image above?

I have installed the trial version and run some tests on Windows with a D415 but was not able to recreate your Nvidia failure, as the GPU inside my NUC8i5INB is Radeon 540x and I encountered no problems.  

There is another sample called 'cpp-realsense' that shows an RGB video with the skeleton overlaid.  Do you encounter any problems with that graphics-based sample too please?

korzen, 2020-02-28:
Thanks for a quick reply the cpp-realsense works fine on CPU but when I run with the GPU I get similar error:
C:\Program Files\Cubemos\SkeletonTracking\bin>cpp-realsense.exe GPU

[CUBEMOS_LOG_1][2020-Feb-28 22:28:38.166043][info]File logging initialised and the files will be saved at the path: "C:\Users\korze\AppData\Local\Cubemos\SkeletonTracking\logs"

[CUBEMOS_LOG_2][2020-Feb-28 22:28:38.176041][info]Cubemos Logging initialised

[CUBEMOS_LOG_3][2020-Feb-28 22:28:38.703872][info]The provided activation key has been accepted for creation of the cubemos handle.

[CUBEMOS_LOG_4][2020-Feb-28 22:28:38.708871][info]Loaded the cubemos plugin C:\Program Files\Cubemos\SkeletonTracking\bin\cubemos_intel_inference_engine_plugin.dll

[CUBEMOS_LOG_5][2020-Feb-28 22:28:40.461313][error]Exception caught in file d:\sdk\skeleton_tracking\dep\cubemos_core\sources_and_scripts\core\modules\engine\src\engine.cpp and line 244 with error message: "Unknown exception"
Model loading failed.^C`

MartyG-RealSense, 2020-02-28:
I replicated your test, launching the program from the command prompt with **cpp-realsense.exe GPU** and got the same error as you on my NUC with a non-Nvidia Radeon GPU.  And like your test, it worked fine with  **cpp-realsense.exe CPU** 

***********

[CUBEMOS_LOG_5][2020-Feb-28 22:13:45.304174][error]Exception caught in file d:\sdk\skeleton_tracking\dep\cubemos_core\sources_and_scripts\core\modules\engine\src\engine.cpp and line 244 with error message: "Unknown exception"
Model loading failed.

***********

The error seemed to be particular to examples that use graphics.  I tried the GPU instruction with **csharp-interactive.exe GPU** though and did not get the error.

[image]

korzen, 2020-02-29:
That's because the csharp-interactive.exe does not recognize GPU option. See console log in the attached image. The inference runs on CPU.

closed 2020-03-04 by MartyG-RealSense

Maurycy-Krzeminski, 2021-07-13:
Is tere a way for it to work on 11th Gen Intel(R) Core(TM) i7-11370H @ 3.30GHz  
Cause when i try to do use it on python "A compatible CPU 
device was not found. Use the cubemos SDK with a compatible system as listed in the system requirements.. Return code: 1"

MartyG-RealSense, 2021-07-13:
Hi @Maurycy-Krzeminski   The Cubemos company are responsible for technical support for the **Cubemos Skeleton Tracking SDK**, as it is not an Intel product.  You can submit a question to Cubemos by visiting the site below and clicking on the **Contact Us** button.

https://www.cubemos.com/skeleton-tracking-sdk

OlalekanIsola, 2021-08-03:
Hi, @MartyG-RealSense  I have been able to install, run the cubemos and everything is fine. However, when I edit or try to save the skeleton coordinates in the python file(**skeleton-tracking-realsense.py**), I keep getting this error below. I am currently working with the trial license. Kindly help

```
Cannot save C:\Program Files\Cubemos\SkeletonTracking\samples\python\basics.py.
Unable to create a backup file (basics.py~).
The file left unchanged.
```

MartyG-RealSense, 2021-08-03:
Hi @OlalekanIsola  The file should be saved to a path that permits the writing of files (i.e the folder is not a protected read-only one).  Below are some example commands:

**Via Cmd**
python skeleton-tracking-image.py -o C:\tmp\output.jpg ^
..\res\images\skeleton_estimation.jpg

**Via Python**
python3 skeleton-tracking-image.py -o ~/output.jpg
../res/images/skeleton_estimation.jpg

If you are not able to save to a location that you know should be writable then please contact Cubemos support for further assistance using the Contact Us button on the Cubemos Skeleton Tracking SDK website at the link below, as Cubemos handle technical support for this product.  Thanks!

https://www.cubemos.com/skeleton-tracking-sdk

OlalekanIsola, 2021-08-03:
Thank you so much @MartyG-RealSense, However, I am not trying to save the image, I am trying to save the coordinates(x,y, z) continuously. I will like to save every skeleton coordinates. ie each joint coordinate is saved in CSV files as long I keep recording

MartyG-RealSense, 2021-08-03:
I recommend asking Cubemos tech support.  I do apologize.

OlalekanIsola, 2021-08-03:
Thank you @MartyG-RealSense, I have sent an email to them. Thank you for your help

alale70, 2022-08-07:
Hi @MartyG-RealSense,
I have some questions:
1. does cubemos have output in excel? I need the key points and their coordinates in an excel or CSV file.
2.  where is the origin (0,0,0)  in the webcam skeleton tracking?
3. are the coordinates in metric space or not?
4.  what is the third value in the presented coordinates of the skeleton (x,y,?)?
I really appreciate your help

alale70, 2022-08-07:
> Dear @OlalekanIsola 
can you solve your problem to save the coordinates in a CSV file?

MartyG-RealSense, 2022-08-07:
Hi @alale70  I can only point you towards the Cubemos company's official documentation for the Cubemos Skeleton Tracking SDK as this support forum cannot provide assistance with this software.

**'Getting Started' User Guide PDF**

https://download-skeleton-tracking-sdk.s3.eu-central-1.amazonaws.com/GettingStartedGuide.pdf

Within the guide, it provides the following information about developer documentation.

************

**Developer Documentation**

After installation you find a detailed description of the Skeleton Tracking API in the developer documentation provided in the following path

%CUBEMOS_SKEL_SDK%\docs\doc_doxygen\html\index.html

The developer documentation covers the following content:

**General**

cubemos Skeleton Tracking API

**C++**

Skeleton Tracking on the webcam or video input - C++
Skeleton Tracking on Intel® RealSense™ Streams - C++
Skeleton Tracking on a single image - C++

**C#**

cubemos Skeleton Tracking API C# Bindings
Skeleton Estimation on a local image - C#
Skeleton Estimation on Intel® RealSense™ RGB Image - C#
Skeleton Tracking on a sequence of Images read from a disk- C#
Skeleton Tracking on Intel® RealSense™ Streams with 3D - C#

**Python**

cubemos Skeleton Tracking API Python Bindings
Skeleton Estimation on a single image – Python
Skeleton Tracking on a webcam or video input – Python
Skeleton Tracking on Intel® RealSense™ Streams with 3D – Python