realsenseai/librealsense #5941: KinectFusion in the browser with realsense and webGL2-compute

philipNoonan, 2020-02-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17)  | 
| Platform                                 | PC  |

Hi all, hope this is ok to share here, please remove if you like, but I hope it will be of interest to this community.

As a side project I was keen on looking into using the realsense inside a browser but didnt have much luck getting the nodeJS samples to work in this repo or the demo's here https://github.com/intel/depth-camera-web-demo as the w3c spec seemed to have dropped a lot of the support which that code relied on (such as the videoKind: depth constraints).

Fortunately accessing the depth camera as a normal userMedia device still works great, so I hacked together some stuff and put together a simple demo showcasing a KinectFusion implementation which utilises the new webGL2-Compute capability of Chrome (> v80).

You need to set two flags in chrome (goto chrome://flags) to enable webGL2.0-Compute which are
`WebGL 2.0 Compute` Enabled
`Choose ANGLE graphics backend` OpenGL 

After that you can try the code here, on page load, you can press 'Load Realsense' to start the camera, and then by clicking on any depth surface on the depth image you can re-center the KinectFusion tracking volume. Resetting the volume can be acheived by pressing the 'Reset Volume' button.

[https://philipnoonan.github.io/wglFusion/depthColorDemo.html](url)

I've tested this mainly in windows, but it does work in ubuntu, but I think you need Chrome Dev rather than the offical build. I have tested on various systems with GTX 1070, 1080, 2080, various quadro's, and a bootcamped windows 10 running on a 2019 Macbook Pro. Note this will not run on MacOS due to Apple's lack of support for OpenGL.

Hopefully this will be of interest to anyone wanting to see what you can do with the realsenses in the browser :)

MartyG-RealSense, 2020-03-02:
Thanks so much for your great work on this project!  I will promote a link to it on Intel Support's RealSense forum too.

MartyG-RealSense, 2020-03-12:
I will close this case now.   It will still be accessible through browser searches.  Thanks again for your work on this project!

closed 2020-03-12 by MartyG-RealSense