realsenseai/librealsense #5942: Kernel Patches Ubuntu

d3-worgan, 2020-02-28:
Hi can anyone advise how to continue in this situation?  Appears to be similar to the one described here https://github.com/IntelRealSense/librealsense/issues/5748 i.e. have followed the installed steps described at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md and am now stuck on running step 4, "./scripts/patch-realsense-ubuntu-lts.sh"

[image]

Thanks
Dan

_Originally posted by @d3-worgan in https://github.com/IntelRealSense/librealsense/issues/5748#issuecomment-592479896_

christian-rauch, 2020-02-28:
> Hi can anyone advise how to continue in this situation? Appears to be similar to the one described here i.e. have followed the installed steps described at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md and am now stuck on running step 4, "./scripts/patch-realsense-ubuntu-lts.sh"
> 
> [Screenshot from 2020-02-28 11-34-28crop]
> 
> Thanks
> Dan

Rule Nr. 1: Never paste screenshots of text. Past the text directly as code.

> Hi can anyone advise how to continue in this situation? Appears to be similar to the one described here #5748 i.e. have followed the installed steps described at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md and am now stuck on running step 4, "./scripts/patch-realsense-ubuntu-lts.sh"
> 
> [Screenshot from 2020-02-28 11-34-28crop]
> 
> Thanks
> Dan
> 
> _Originally posted by @d3-worgan in [#5748 (comment)](https://github.com/IntelRealSense/librealsense/issues/5748#issuecomment-592479896)_

Rule Nr. 1: Never paste screenshots of text. Paste the text directly as formatted code.

I do not see an error in this screenshot. The compilation seems to succeed.
Is it just the warning that is confusing? The kernel 5.3 series is supported since https://github.com/IntelRealSense/librealsense/pull/5751.

d3-worgan, 2020-02-28:
Thanks for the reply Christian.

Yes was just worried it said unsupported kernel version for 5.3.0 and that is the kernel I am using. I can see that 5.3.0 has been supported for a few weeks now which is why it appeared confusing.

Just wanted to check everything is okay so thanks Christain

Dan

christian-rauch, 2020-02-28:
@ev-mp Can you update the kernel version check?

ev-mp, 2020-03-01:
@d3-worgan , the modified script for kernels 5.0/3 is available on the `development` branch.
The `master` branch is yet to be updated. 

Please check and confirm.

ev-mp, 2020-03-08:
@d3-worgan , the `master` branch has been up-streamed with that fix last Monday, so you can use either one to apply the patches.
Actually, I would recommend to reconsider the deployment scheme and switch from scripts  to DKMS, mainly due to more efficient maintenance model.

closed 2020-03-16 by ev-mp