realsenseai/librealsense #5943: depth array (ushort) and getDistance of (x,y) are not matching

priyanandanaren, 2020-02-28:
Depth array (ushort) value (number*depthscale) and depthframe's getDistance values are not matching. For example, the pixel point (x,y) depth is not matching with the depth array (say x*1280+y). How to map the x,y and ushort array index properly?

RealSenseCustomerSupport, 2020-03-02:
----------------------------------------------

Hi @priyanandanaren,

Can you please elaborate what are you trying to achieve, what language do you use and provide some numerical examples?

Thank you!

priyanandanaren, 2020-03-03:
it is working now. I made some correction and it is ok now.

RealSenseCustomerSupport, 2020-03-03:
----------------------------------------------

@priyanandanaren,

Glad to know it is working now! Please let us know if you need any further assistance with this matter or if the issue can be closed.

Thank you!

closed 2020-03-05 by dorodnic