realsenseai/librealsense #5944: Mapping color X, Y  coordinates in to  depth x, y coordinates

XMaster96, 2020-02-29:
Hello,
I am performing object detection on the color Image, what then gives me x, y coordinates in the color Image space. I now would like to map the color x, y coordinates in to the depth x, y coordinate space.
Is there a way to do that with the SDK or to retrieve the transformation Matrix?, The Depth date and Image have the same size (1280x720), and i am using the D435i and Python3.

Thanks

EDIT:
This question is similar to this one:
https://github.com/IntelRealSense/librealsense/issues/5943#issue-572912072

MartyG-RealSense, 2020-02-29:
If you need to map color pixels to depth pixels, I wonder if the instruction **rs2_project_color_pixel_to_depth_pixel** may meet your needs.

https://github.com/IntelRealSense/librealsense/issues/2948

RealSenseCustomerSupport, 2020-03-20:
----------------------------------------------

@XMaster96 Did you get chance to try what MartG-RealSense suggested? Any update? Thanks!

RealSenseCustomerSupport, 2020-03-27:
----------------------------------------------

@XMaster96 Any update from your side? Thanks!

XMaster96, 2020-03-27:
Sorry for not answering.
The priorities in my project have changed, and so I haven't tried it yet. I looked in to the link from @MartyG-RealSense, and it seems to solve my problem
I am also closing this issue for now, because I do not know when I will come back to it.

Thanks a lot for the the help

closed 2020-03-27 by XMaster96