realsenseai/librealsense #5945: T265 not work

dongxuanlb, 2020-03-01:
Use rs-multicam example:

RealSense error calling rs2_pipeline_start_with_config(pipe:0x55d4cd235d60, config:0x55d4cd23cca0):
    No device connected

FW: 0.2.0.926.

but the realsense-view is ok.

MartyG-RealSense, 2020-03-01:
Can rs-multicam detect the T265 if you unplug it from the USB port and then plug it back in, please?

dongxuanlb, 2020-03-01:
> Can rs-multicam detect the T265 if you unplug it from the USB port and then plug it back in, please?

No, I had try many times. Actually, use ctx_.query_devices() can found the t265, every info is ok. But when call cfg.resovle, or just start pipeline it throws the message "No device connected"

MartyG-RealSense, 2020-03-01:
Do you have any other RealSense cameras plugged in as well as the T265 when using rs-multicam, please?

dongxuanlb, 2020-03-01:
> Do you have any other RealSense cameras plugged in as well as the T265 when using rs-multicam, please?

No, I only use T265 and the d435i  is ok.

and I look up the realsense-view cpp and use the code is working correctly.

for(auto && s : device_.query_sensors()) {
    if (s.is<rs2::pose_sensor>()) {
        rs2::pose_sensor ps = s.as<rs2::pose_sensor>();
        std::vector<rs2::stream_profile> sps = ps.get_stream_profiles();
        for(auto && sp : sps) {
            if (sp.is<rs2::pose_stream_profile>()) {
                ps.open(sp);
            }
        }
        ps.start([&](const rs2::frame& frame) {
            RCLCPP_INFO(get_logger(), "get");
        });
    }
}

MartyG-RealSense, 2020-03-01:
I am not a RealSense programming specialist.  Others on the RealSense team can provide more expert answers about coding.  From what I understand though,  using query_devices() is the appropriate action when starting a T265, so it is understandable that this would work.

Things become more complicated when using a T265 and a depth camera such as D435 together on the same machine.  In that case, you also have to distinguish between the devices by providing a means to specially identify the T265.  This may be with a serial number, or by querying for devices whose name includes the word 'T265'.   You should not need to have a special ID if you are using the T265 on its own though with no other devices attached to the computer (rs-multicam checks for every RealSense device that is plugged in when the program starts).

MartyG-RealSense, 2020-03-01:
If I may also ask, do you intend to use the T265 and D435i together?  If you only plan to use the T265 on its own with no other camera attached, then you could also use the **rs-pose** example to get pose data, or the **rs-ar-basic** example to get pose and fisheye together.

https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#c-examples

dongxuanlb, 2020-03-01:
Yes, I have a d435i and I know that use d43x and t265 work together would cause a lot of little problem. since the latest sdk version 2.32.1, I test all functions I needs with my d435i , and work correctly .

MartyG-RealSense, 2020-03-01:
SDK 2.32.1 did a massive rewrite of how the T265 is handled, and caused problems for a small number of people that were not in the previous 2.31.1 version.  The RealSense team are working on fixes for issues that arose from the changes in 2.32.1.

neilyoung, 2020-03-02:
@dongxuanlb Would you mind to checkout my little python scripts? 
https://github.com/neilyoung/t265pose/releases/tag/v0.2.0

smartynenko, 2020-03-05:
same error when I ran rs_pose example. It was working fine when I built out of dev branch prior to 2.33.1 merge (I'm not sure if it is still will work if you build off dev branch)

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

There was an update to the latest 2.33.1 librealsense that deals with the serial number compatibility for T265

https://github.com/IntelRealSense/librealsense/pull/5645

https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1

Review this latest update utilizing serial numbers and see if it works with your configuration.

tdudz, 2020-03-06:
@RealSenseCustomerSupport 

i just tried the new v2.33.0 release. I still get the same error, "no device connected" when trying to run two T265s together. Had the same issue with v2.32.1. Is there a different way to connect and initialize the T265 devices? Could a new code example be provided?

For the time being, I have downgraded back to v3.21.0 since my code runs fine there. Please advise!

FLS-BP-US, 2020-03-11:
I believe I am having a similar problem.  I am running with pyrealsense2 version 2.33.1.1370.  If I have both the D435I and the T265 plugged in, I can only see the D435I.  When I unplug the D435I, I can see the T265.

RealSenseSupport, 2020-09-11:
Thank you for highlighting the continued issue of using T265 with other RealSense products. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

neilyoung, 2020-09-11:
@RealSenseCustomerSupport Could you please elaborate on this statement? Is the T265 retired? Will you no longer support this camera in future?

RealSenseSupport, 2020-09-15:
T265 is not discontinued or retired, and is available. We have moved our focus to our next generation of products and consequently, we will not be addressing issues/new features in the T265.

neilyoung, 2020-09-16:
@RealSenseSupport Thanks. Maybe I got you wrong, so please explain: It is all about what you understand as "product". If it is the camera, then I would ask for a successor of the T265, which does not have this issue.

If you mean your SDK and firmware by "product", and with your statement, that you are not fixing issues in older versions of the SDK and this issue will be or is already fixed in later versions of the SDK and/or firmware, then I would be with you and would totally understand that. 

So I'm sorry, I would need to ask you again for clarification. If it is none of the above and you are simply rejecting to care for your product T265, which is still on sale, then I would not understand such a policy shown by a company like Intel.

smartynenko, 2020-09-16:
@neilyoung - it looks like they are phasing our the chip that 266 is based on. And unless Intel will provide us with guidance on the future direction, we can only speculate about future of the product 

https://ark.intel.com/content/www/us/en/ark/products/122461/intel-movidius-myriad-2-vision-processing-unit-4gb.html

neilyoung, 2020-09-16:
@smartynenko Thanks for the link. Makes sense. So basically don't bet on the T265 anymore... Not future proof.

"T265 is not discontinued "

Really?

smartynenko, 2020-09-16:
@RealSenseSupport - could you provide forward guidance on expected lifecycle of T265? Is there a successor planned and will be announced in the near future, or T265 was a one-off kind of product and will be phased out once current stock is sold out?

RealSenseSupport, 2020-09-23:
T265 is not discontinued or retired, and is available. We have moved our focus to our next generation of products and consequently, we will not be addressing issues/new features in the T265.

Cannot speak about products under development that have not been released yet.  Please continue to check our site for product releases and current available product:  https://www.intelrealsense.com/

kenfast, 2020-11-23:
@RealSenseSupport, so the T265 is "unsupported", even though it is not "discontinued, retired, or unavailable?"  And by not addressing issues, you include the issue that it cannot connect reliably to many Windows 10 machines?  I guess it is just too bad for those of us who are trying to use this for real work.  That seems pretty sloppy for a big-name company like Intel.

closed 2024-04-01 by MartyG-RealSense