realsenseai/librealsense #5947: Cubemos skeletal tracking sdk installation failed - ubuntu

devshank3, 2020-03-01:
# cubemos skeletal tracking 
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   |  | 
| Operating System & Version |    Linux (Ubuntu 19.10)  | 
| Kernel Version (Linux Only)    |   5.3.0-40-generic                                       | 
| Platform                                 | PC |
| SDK Version                            |                           | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  others                         | 

### Issue Description

Cubemos Installation 

I folllowed this link for installation 

https://dev.intelrealsense.com/docs/skeleton-tracking-sdk-installation-guide#section-3-linux-installation-and-license-activation

but i get this issue 

[image]

any one successfully installed on ubuntu

closed 2020-03-01 by devshank3

feimeng678, 2020-04-03:
I encounter the same one. But my OS is ubuntu16.04, while the instruction shows it is feasible on 18.04.

OlalekanIsola, 2021-01-17:
I get this same error on Ubuntu 16.04. Is there any possible fix  this please

andDecarlo, 2021-03-24:
I am getting the same issue on 20.04 with libboost-1.71 installed.

fse5947, 2021-08-04:
Same issue here as well on 18.04 with libboost installed. Why was this issue closed without being resolved? Has anyone had any luck with it?

devshank3, 2021-08-07:
I closed this issue because, My Laptop's CPU unit was using AMD Ryzen 3550H and I guess...... Cubemos products are not supported for AMD chips.

fse5947, 2021-08-07:
I have an Intel chipset CPU so I think it has bigger issues than that. Tried in on a Windows OS and didn't work there either giving me a "failed to receive frames" error.

OliviaNocentini, 2021-09-19:
Hi all, is there any plan to support also ubuntu 20.04?