realsenseai/librealsense #5949: Announcement:  RealSense SDK version 2.33.1 and new firmware 5.12.3.0 now available

MartyG-RealSense, 2020-03-02:
Hi everyone,

RealSense SDK version 2.33.1 and a new firmware for the 400 Series cameras (5.12.3.0) is now available.  Firmware 5.12.3.0 contains stability and power management fixes and enhancements.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1 

**API CHANGES**

The extensive list of API changes can be viewed here:

https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2331 

**SDK ENHANCEMENTS**

The list of RealSense SDK enhancements is also extensive and can be viewed here:

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2331

kafan1986, 2020-03-02:
Can't find the 5.12.3 firmware

MartyG-RealSense, 2020-03-02:
If you have installed SDK version 2.33.1 and can access the RealSense Viewer then you should be able to update a camera to it by going to the More icon at the top of the options side-panel and selecting the menu option to update to recommended firmware (the new firmware is bundled inside the Viewer).  I did this update with a D435i earlier today.

kafan1986, 2020-03-02:
I don't have a windows based system. I update through my own Android app. So need the firmware file to download first.

MartyG-RealSense, 2020-03-02:
A member of the RealSense team should add the new firmware as a downloadable file to the firmware release page at some point, though I do not know when exactly.

https://dev.intelrealsense.com/docs/firmware-releases

The sentence "stability and power management fixes and enhancements" is exactly how a RealSense team member described the additions in the new firmware.

ev-mp, 2020-03-03:
@kafan hello, while the developer's zone section being updated you can get the new D4xx firmware directly from http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/RS4xx/FW/D4XX_RC_Image-5.12.3.0.bin.
Actually the link is not really necessary as you can get the that FW (along with SR300 and T265) image(s) just by invoking the SDK's CMake - the FWs are bundled with the SDK build and will be downloaded automatically.

closed 2020-03-04 by MartyG-RealSense