realsenseai/librealsense #5950: Restore depth from 2 infrared images

Allius27, 2020-03-02:
| | |
|---|---|
|**librealsense**|2.32.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|915112071095|
|**Firmware Version**|05.12.02.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|901523021687|
|**Firmware Update Id**|901523021687|

Good day.

I found, that realtime saving filtered depth frame in range 0.35 to 0.7 meters (with function "export_to_ply") provides a load for processor up to 75% and has used hard disk with speed 50 Mb/s

I need to store data from camera for further analize.
I thought that I can save both infrared images separetly and when reconstruture depth from them. 

So, does librealsense can restore depth image from two infrared images?

MartyG-RealSense, 2020-03-02:
I do not have a complete answer to this question - the other Intel RealSense team members on this forum may be able to help.  I will try to provide a starting point though.

If you want to combine left and right IR data sources manually to create your own depth frame instead of letting the RealSense hardware create the depth frame, I believe that the appropriate course of action would be to use **software device**.  This enables you to create synthetic frames in a way that does not require the RealSense hardware.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

An alternative to software device may be a **custom processing block**, as discussed recently in the conversation linked to below.

https://github.com/IntelRealSense/librealsense/issues/5847

dorodnic, 2020-03-03:
Hi @Allius27 
Depth calculation is done on chip and not in software, so there is no way to reconstruct depth from IR pair using the SDK

Allius27, 2020-03-03:
Sergey, thank for answer 

Do you have some plans for adding that feature in next releases?

dorodnic, 2020-03-03:
D4 ASIC depth algorithm is part of the hardware offering and we are not planning to publish software version of it.
If you are capturing rectified left and right infrared images (format Y8), you can fairly easily feed these into OpenCV built-in depth engine (StereoBM), but the output will be different from D4 depth.

closed 2020-03-03 by Allius27