realsenseai/librealsense #5951: Regarding Intel® RealSense™ SDK 2.0 (build 2.33.1)

neilyoung, 2020-03-02:
There is a new release announced here
https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1

I'm either too stupid or no longer used to follow the latest twist of Intel Realsense, but I'm completely unable to see that.

It is not on the master branch, but on `rc`, which is fine. But I checked out `rc` compiled from scratch on macOS, installed it all.

```
~/Documents/librealsense/build $ git branch
  master
* rc

~/Documents/librealsense/build $ git log --oneline
842ee1e1e (HEAD -> rc, tag: v2.33.1, origin/rc) Merge pull request #5948 from ev-mp/master
2e0a03ef7 Merge pull request #5946 from ev-mp/rc
9a2f2cffd Promote D4xx firmware version to 5.12.3.0
299a524d1 Merge pull request #5923 from dorodnic/rc
032b7585b (origin/master, origin/HEAD, master) Comment on the wrappers
9c845f56d Rectify license of the INF file
06441e535 Reverting merge mistake
4d7c0404c Merge pull request #5900 from ev-mp/rc
04d3019fb Bump version to v2.33.1
da66c7cb4 Fix memory leak in v4l2 backend on kernel 
```

I think that I'm on the right branch and commit, but whatever I do: My T.265 always reports to see SDK 2.32.1 with FW 0.2.0.926. 

What am I doing wrong?

dorodnic, 2020-03-04:
Hi @neilyoung 
There is nothing special about the rc branch. We also completed the release by merging to master as usual. I suspect you have another version installed, did you try `sudo make uninstall` from the build directory?

neilyoung, 2020-03-04:
@dorodnic Yepp. Did that before and cloned the entire project before making. 

Will repeat now and report

neilyoung, 2020-03-04:
BTW: While uninstalling right now I found this:

-- Uninstalling "/usr/local/lib/pyrealsense2.2.33.1.cpython-37m-darwin.so"
-- Uninstalling "/usr/local/lib/pyrealsense2.2.33.cpython-37m-darwin.so"
-- Uninstalling "/usr/local/lib/pyrealsense2.cpython-37m-darwin.so"
-- Uninstalling "/usr/local/lib/pybackend2.2.33.1.cpython-37m-darwin.so"
-- Uninstalling "/usr/local/lib/pybackend2.2.cpython-37m-darwin.so"
-- Uninstalling "/usr/local/lib/pybackend2.cpython-37m-darwin.so"
-- Uninstalling "/usr/local/lib/libuvc_fw.a"

So there is (was) a 2.33 version somehow... Strange. realsense-viewer and the SDK did report 2.32.1

neilyoung, 2020-03-04:
Yepp, thanks. My bad. Everything OK.

closed 2020-03-04 by neilyoung