realsenseai/librealsense #5952: Auto Exposure prevents manual gain setting

stephentown42, 2020-03-02:
| | |
|---|---|
|**librealsense**|2.31.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|733512070568|
|**Firmware Version**|05.12.01.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|735413020246|
|**Firmware Update Id**|735413020246|

Received error when trying to manually change Gain on IR cameras with auto exposure on:

_"Backend in rs2_set_option(options:000001851211BB20, option:Gain, value:100):
hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."_

Resolved by turning auto-exposure off, changing gain successfully and re-engaging autoexposure. Would be helpful to have instructions in error message that gain cannot be changed manually without turning autoexposure off.

MartyG-RealSense, 2020-03-02:
Thank you very much for your suggestion.   I will tag it as a feature request so that the RealSense developer team can officially track it.

ev-mp, 2020-05-07:
@stephen-systemfriend , in the next version Gain control for Depth sensor will override and disable Auto-exposure in the same manner as Manual Exposure control does.
https://github.com/IntelRealSense/librealsense/pull/6355

stephen-systemfriend, 2020-05-07:
I think you meant @stephentown42

MartyG-RealSense, 2020-06-03:
Hi @stephentown42   The fix referenced above - https://github.com/IntelRealSense/librealsense/issues/5952#issuecomment-625044300 - has now been merged into the RealSense SDK in version **2.35.0**.

This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thanks!

closed 2020-06-09 by MartyG-RealSense