realsenseai/librealsense #5953: Error with attempting to get aligned frame through syncer

Allius27, 2020-03-02:
| | |
|---|---|
|**librealsense**|2.32.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|915112071095|
|**Firmware Version**|05.12.02.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|901523021687|
|**Firmware Update Id**|901523021687|

Good day.

When I am trying to get aligned frame to depth through rs2::syncer, I get error `Error occured during execution of the processing block! See the log for more info`

What I am doing wrong?

Program will work if string 
```
 rs2::align align_to_depth(RS2_STREAM_DEPTH);
```
will change  to 
```
 rs2::align align_to_depth(RS2_STREAM_COLOR);
```

I slightly modified code from [issue 1984] to demonstrate problem (https://github.com/IntelRealSense/librealsense/issues/1984)

So, here my code 

```
#include <librealsense2/rs.hpp>
#include <vector>
#include <iostream>

#include <opencv2/highgui.hpp>
#include <opencv2/core.hpp>

using namespace std;
using namespace rs2;

enum SENSOR{
 DEPTH,
 COLOR
};

stream_profile getStreamProfile(sensor s, int w, int h, int fps, rs2_format format)
{
 for (auto p : s.get_stream_profiles())
 {
  if (p.as<video_stream_profile>().width() == w &&
          p.as<video_stream_profile>().height() == h &&
          p.fps() == fps && p.format() == format)
      return p;
 }
}

int main(int argc, char * argv[]) try
{
 rs2::colorizer color_map;
 vector<stream_profile> sps;

 rs2::context ctx;
 syncer sync;
 device dev = ctx.query_devices()[0];
 vector<rs2::sensor> sensors = dev.query_sensors();

 rs2::align align_to_depth(RS2_STREAM_DEPTH);

 sps.push_back(getStreamProfile(sensors[DEPTH], 1280, 720, 30, RS2_FORMAT_Z16));
 sps.push_back(getStreamProfile(sensors[COLOR], 1280, 720, 30, RS2_FORMAT_RGB8));

 sensors[DEPTH].open(sps[DEPTH]);
 sensors[COLOR].open(sps[COLOR]);

 sensors[DEPTH].start(sync);
 sensors[COLOR].start(sync);

 while (true)
 {
  rs2::frameset data = sync.wait_for_frames();

  data = align_to_depth.process(data);
  rs2::video_frame color = data.get_color_frame();

  if (!color)
      continue;

  cv::Mat depthSetColorMat = cv::Mat( color.get_height(), color.get_width(),
                                   CV_8UC3, (void*)(color.get_data()) );

  cv::imshow("depth aligned", depthSetColorMat);
  cv::waitKey(1);
 }
  return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

dorodnic, 2020-03-05:
Hi @Allius27 
I think that you need to add a check before:
```cpp
rs2::frameset data = sync.wait_for_frames();
```
To make sure only frame-sets of size 2 are passed to alignment. 
Syncer does not "pad" missing pairs with old frames, unlike pipeline.

Allius27, 2020-03-05:
Thank you

One more question. 
How I can get aligned color image ( with borders ), with turned off depth?
Does librealsense have this functional?

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

The alignment block requires depth stream to work.
Please refer to the following link.
https://github.com/IntelRealSense/librealsense/issues/1556#issuecomment-389061477
Please feel free to open a new ticket if there is any other question.
We will close this case after 7 days from now unless otherwise. Thank you.

closed 2020-03-28 by RealSenseCustomerSupport