realsenseai/librealsense #5955: Realsense-viewer D435i  & Kernel 4.15 warnings - uvc metadata issues

jorgemia, 2020-03-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | V05.12.02.100 | 
| Operating System & Version | Linux Ubuntu 16.04.6  | 
| Kernel Version (Linux Only)    |  4.15.0-37-generic for UP boards (https://wiki.up-community.org/Ubuntu)  | 
| Platform                                 | UP Squared (https://up-board.org/upsquared/specifications/) |
| SDK Version                            |   2.31.0                           | 
| Segment			|  Robot                           | 

### Issue Description
I've found similar issues (#5815) but can't seem to solve the problems I'm getting.

I have an UP Squared Board running Ubuntu 16.04 and Kernel 4.15 made for UP boards (which I installed through the UP Board Wiki as the recommended Kernel to run https://wiki.up-community.org/Ubuntu). I installed librealsense 2.31 from source as I intend to use ros-realsense soon. When I launch realsense-viewer I get the following warnings:

**Warning 1:**
`WARNING [140520726722304] (types.cpp:49) hwmon command 0x4f failed. Error type: No data to return (-21).`

If I understand correctly, this means my camera's IMU hasn't been calibrated?

**Warning 2:** 
`[140520726722304] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!`

**Warning 3:**
`[140520575719168] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.`

This happens when I turn on either the stereo or RGB module. I installed librealsense from source and patched the kernel using ./scripts/patch-realsense-ubuntu-lts.sh. Why am I getting this error and how can I solve it?

**Warning 4:**
`WARNING [140520575719168] (sensor.cpp:962) HID timestamp not found, switching to Host timestamps.`

This happens when I turn on the motion module. Is this a problem?

I would like to resolve these issues before installing ros-realsense so any help would be greatly appreciated. Thanks!

MartyG-RealSense, 2020-03-02:
Error type: No data to return (-21) indicates an uncalibrated IMU, as you suggest.

If you experience problems with the normal installation process, there is a "bypass" method of installation if you have an internet connection.  This enables installation on most forms of Linux without needing to be concerned with Linux types, patching and kernel versions.  Instructions for doing the install on SDK version 2.30.0 or newer can be found in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360037657234/comments/360009799533

ev-mp, 2020-03-03:
@jorgemia , the link referenced above above is the Vendor's guide for OS installation.
Once accomplished, you'll need to follow the [SDK installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution). More specifically, in case you need an access to metadata attributes the you'll have to patch the kernel via [script](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md):
```sh
./scripts/patch-realsense-ubuntu-lts.sh
```

,or with [DKMS](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) Debian package:
```sh
sudo apt-get install librealsense2-dkms
```

jorgemia, 2020-03-03:
@ev-mp I have done that (via script) yet the errors still appear. I also tried uninstalling via script and installing everything through Debian yet I still get the UVC metadata warning.

When running `modinfo uvcvideo | grep "version:" ` I don't see realsense anywhere so it seems the kernel is not being updated.

Why could that be?

@gbr1 suggested installing linux-headers before installing librealsense. I also tried that yet I'm still getting the warnings when running realsense-viewer.

ev-mp, 2020-03-03:
@jorgemia, then apparently the patch was not applied properly.
The kernel version seem to be of the supported LTS-type, still there could be UpBoard-specific attributes that differenciate it from the stock LTS kernel.
Can you share the log produced by running the script in the terminal?  Having the kernel log (`dmesg -wT`) would be useful as well.

jorgemia, 2020-03-03:
This is what I get when I run `./scripts/patch-realsense-ubuntu-lts.sh`:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.06.95-9build1).
build-essential is already the newest version (12.1ubuntu2).
git is already the newest version (1:2.7.4-0ubuntu1.7).
linux-headers-generic is already the newest version (4.4.0.174.182).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-xenial-hwe folder

Initialised empty Git repository in /home/jorge/librealsense-2.31.0/ubuntu-xenial-hwe/.git/
Fetching Ubuntu LTS tag Ubuntu-hwe-4.15.0-37.40_16.04.1  to the local kernel sources folder
remote: Counting objects: 5820100, done.
remote: Compressing objects: 100% (877345/877345), done.
remote: Total 5820100 (delta 4934074), reused 5784079 (delta 4899700)
Receiving objects: 100% (5820100/5820100), 1.23 GiB | 1.30 MiB/s, done.
Resolving deltas: 100% (4934074/4934074), done.
From git://kernel.ubuntu.com/ubuntu/ubuntu-xenial
 * [new tag]         Ubuntu-hwe-4.15.0-37.40_16.04.1 -> Ubuntu-hwe-4.15.0-37.40_16.04.1
Switching to LTS tag Ubuntu-hwe-4.15.0-37.40_16.04.1
Checking out files: 100% (64439/64439), done.
Note: checking out 'Ubuntu-hwe-4.15.0-37.40_16.04.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at f42c0ad... UBUNTU: Ubuntu-hwe-4.15.0-37.40~16.04.1
Applying patches for xenial-hwe line
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file drivers/media/v4l2-core/v4l2-ioctl.c
patching file include/uapi/linux/videodev2.h
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 2786 (offset 15 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Hunk #1 succeeded at 1226 (offset 4 lines).
Hunk #2 succeeded at 1264 (offset 4 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 199 (offset 12 lines).
Hunk #2 succeeded at 220 (offset 12 lines).
Applying realsense-hid patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
patching file drivers/iio/gyro/hid-sensor-gyro-3d.c
Applying realsense-powerlinefrequency-fix patch
patching file drivers/media/usb/uvc/uvc_ctrl.c
Retrofit uvc bug fix enabled with 4.18+
patching file drivers/media/usb/uvc/uvc_video.c
Unreversed patch detected!  Ignore -R? [n] n
Apply anyway? [n] n
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/media/usb/uvc/uvc_video.c.rej

```

And this is what I get from `dmesg -wT`

[... 1373 lines cut ...]
[Tue Mar  3 14:03:18 2020] CR2: 00007fcde09ba000 CR3: 000000014b4d6000 CR4: 00000000003406f0
[Tue Mar  3 14:03:18 2020] Call Trace:
[Tue Mar  3 14:03:18 2020]  v4l_enum_fmt+0x75/0x120 [videodev]
[Tue Mar  3 14:03:18 2020]  __video_do_ioctl+0x2a4/0x320 [videodev]
[Tue Mar  3 14:03:18 2020]  ? futex_wait_queue_me+0xd6/0x120
[Tue Mar  3 14:03:18 2020]  ? futex_wait+0x240/0x260
[Tue Mar  3 14:03:18 2020]  ? video_ioctl2+0x20/0x20 [videodev]
[Tue Mar  3 14:03:18 2020]  video_usercopy+0x285/0x630 [videodev]
[Tue Mar  3 14:03:18 2020]  video_ioctl2+0x15/0x20 [videodev]
[Tue Mar  3 14:03:18 2020]  v4l2_ioctl+0xe8/0xf0 [videodev]
[Tue Mar  3 14:03:18 2020]  do_vfs_ioctl+0xa4/0x600
[Tue Mar  3 14:03:18 2020]  ? SyS_futex+0x7f/0x180
[Tue Mar  3 14:03:18 2020]  SyS_ioctl+0x79/0x90
[Tue Mar  3 14:03:18 2020]  do_syscall_64+0x73/0x130
[Tue Mar  3 14:03:18 2020]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
[Tue Mar  3 14:03:18 2020] RIP: 0033:0x7fce01ce5f47
[Tue Mar  3 14:03:18 2020] RSP: 002b:00007ffc5bc44828 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[Tue Mar  3 14:03:18 2020] RAX: ffffffffffffffda RBX: 00007fcdec0035b0 RCX: 00007fce01ce5f47
[Tue Mar  3 14:03:18 2020] RDX: 00007ffc5bc449b0 RSI: 00000000c0405602 RDI: 0000000000000011
[Tue Mar  3 14:03:18 2020] RBP: 00007ffc5bc44dc0 R08: 0000000005bdb260 R09: 0000000000000001
[Tue Mar  3 14:03:18 2020] R10: 5a3136200000000f R11: 0000000000000246 R12: 00007ffc5bc44bb0
[Tue Mar  3 14:03:18 2020] R13: 0000000000000000 R14: 0000000005bdb370 R15: 000000005a313620
[Tue Mar  3 14:03:18 2020] Code: 30 0f 84 b4 01 00 00 3d 4a 4c 32 30 0f 84 97 01 00 00 3d 4d 33 31 30 0f 84 7a 01 00 00 89 c6 48 c7 c7 92 14 c1 c0 e8 7f fc e8 c9 <0f> 0b 80 7b 0c 00 0f 85 e2 f3 ff ff 8b 43 2c 48 c7 c1 39 08 c1 
[Tue Mar  3 14:03:18 2020] ---[ end trace f222e837e3f5d896 ]---
```

ev-mp, 2020-03-03:
@jorgemia, the prompt that you see [has been removed](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-lts.sh#L133-L138) in v2.33.1
>Retrofit uvc bug fix enabled with 4.18+
patching file drivers/media/usb/uvc/uvc_video.c
Unreversed patch detected!  Ignore -R? [n] n
Apply anyway? [n] n
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/media/usb/uvc/uvc_video.c.rej

Please try the latest version from github (master) and update.

jorgemia, 2020-03-03:
Installed the latest version from source and got this when running` ./scripts/patch-realsense-ubuntu-lts.sh`:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.06.95-9build1).
build-essential is already the newest version (12.1ubuntu2).
git is already the newest version (1:2.7.4-0ubuntu1.7).
linux-headers-generic is already the newest version (4.4.0.174.182).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-xenial-hwe folder

Initialised empty Git repository in /home/jorge/librealsense/ubuntu-xenial-hwe/.git/
Fetching Ubuntu LTS tag Ubuntu-hwe-4.15.0-37.40_16.04.1  to the local kernel sources folder
remote: Counting objects: 5820100, done.
remote: Compressing objects: 100% (877345/877345), done.
remote: Total 5820100 (delta 4934074), reused 5784079 (delta 4899700)
Receiving objects: 100% (5820100/5820100), 1.23 GiB | 1.49 MiB/s, done.
Resolving deltas: 100% (4934074/4934074), done.
From git://kernel.ubuntu.com/ubuntu/ubuntu-xenial
 * [new tag]         Ubuntu-hwe-4.15.0-37.40_16.04.1 -> Ubuntu-hwe-4.15.0-37.40_16.04.1
Switching to LTS tag Ubuntu-hwe-4.15.0-37.40_16.04.1
Checking out files: 100% (64439/64439), done.
Note: checking out 'Ubuntu-hwe-4.15.0-37.40_16.04.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at f42c0ad... UBUNTU: Ubuntu-hwe-4.15.0-37.40~16.04.1
Applying patches for xenial-hwe line
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file drivers/media/v4l2-core/v4l2-ioctl.c
patching file include/uapi/linux/videodev2.h
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 2791 (offset 20 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Hunk #1 succeeded at 1226 (offset 4 lines).
Hunk #2 succeeded at 1264 (offset 4 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 202 (offset 15 lines).
Hunk #2 succeeded at 223 (offset 15 lines).
Applying realsense-hid patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
patching file drivers/iio/gyro/hid-sensor-gyro-3d.c
Applying realsense-powerlinefrequency-fix patch
patching file drivers/media/usb/uvc/uvc_ctrl.c
Retrofit UVC bug fix rectified in 4.18+
checking file drivers/media/usb/uvc/uvc_video.c
Hunk #1 succeeded at 1081 (offset 4 lines).
patching file drivers/media/usb/uvc/uvc_video.c
Hunk #1 succeeded at 1081 (offset 4 lines).
[sudo] password for jorge: 
Prepare kernel modules configuration
/home/jorge/librealsense/ubuntu-xenial-hwe
2620$
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
[... 328 lines cut ...]
  LD [M]  /home/jorge/librealsense/ubuntu-xenial-hwe/drivers/media/v4l2-core/videobuf2-core.ko
  LD [M]  /home/jorge/librealsense/ubuntu-xenial-hwe/drivers/media/v4l2-core/videobuf2-dma-sg.ko
  LD [M]  /home/jorge/librealsense/ubuntu-xenial-hwe/drivers/media/v4l2-core/videobuf2-v4l2.ko
  LD [M]  /home/jorge/librealsense/ubuntu-xenial-hwe/drivers/media/v4l2-core/videobuf-vmalloc.ko
  LD [M]  /home/jorge/librealsense/ubuntu-xenial-hwe/drivers/media/v4l2-core/videobuf-dvb.ko
  LD [M]  /home/jorge/librealsense/ubuntu-xenial-hwe/drivers/media/v4l2-core/v4l2-fwnode.ko
  LD [M]  /home/jorge/librealsense/ubuntu-xenial-hwe/drivers/media/v4l2-core/videodev.ko
  LD [M]  /home/jorge/librealsense/ubuntu-xenial-hwe/drivers/media/v4l2-core/videobuf2-vmalloc.ko
make: Leaving directory '/home/jorge/librealsense/ubuntu-xenial-hwe'
Patched kernels modules were created successfully

0
Replacing videodev  -
	Applying the patched module ...  succeeded
Replacing uvcvideo  -
	Applying the patched module ...  succeeded
Replacing hid_sensor_accel_3d  -
Module is resident, unloading ...  succeeded. 
	Applying the patched module ...  succeeded
Replacing hid_sensor_gyro_3d  -
Module is resident, unloading ...  succeeded. 
	Applying the patched module ...  succeeded

Script has completed. Please consult the installation guide for further instruction.
```

jorgemia, 2020-03-03:
Opened realsense-viewer, updated the camera to 5.12.3 and no longer get warnings 2, 3 or 4 in the terminal as described in my issue, so it seems the kernel has been updated. The only one I get is IMU not calibrated (which makes sense as I have not done this). 

Inside realsense viewer, however, the following error now shows up sometimes when starting the motion module (IMU):

Timestamp: 1583252844262.769287
Severity: Error
Description: Motion Module failure

Is this because the IMU is not calibrated?

jorgemia, 2020-03-03:
Calibrated the IMU and get no more warnings on startup of realsense-viewer. Should I close this issue and create a new one for the _motion module failure_, which persists after calibration?

ev-mp, 2020-03-04:
@jorgemia ,  if after receiving the message no IMU data is produced - then yes.
Otherwise you can disregarded it as the SDK provides a w/a to handle it internally

jorgemia, 2020-03-04:
@ev-mp IMU data comes through so won't open an issue for now. Thanks!

One last thing before closing, when I run `modinfo uvcvideo | grep "version:"` I don't see _realsense_ anywhere. Is this something to worry about?

ev-mp, 2020-03-04:
No, that is the expected behavior. The modification of `modinfo uvcvideo | grep "version:` applies only to DKMS package [as stated](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages).

jorgemia, 2020-03-04:
Great, thanks for the help @ev-mp !

closed 2020-03-04 by jorgemia