realsenseai/librealsense #5956: binary package update on jetson

cimbar, 2020-03-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |   Ubuntu 16.04 L4T 28.2  | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  > 28.1                     | 

Hi,
I have been using the binary installed librealsense 28.1 version on Jetson TX2. When i follow the linux or jetson installation pages, i cannot see the newer binary versions of librealsense built for jetson although i updated public keys. I checked versions with `apt-cache showpkg` after `apt-get update` and cannot see newer versions than 28.1. I followed same procedure on my desktop and updated without any problem. Aren't newer versions of librealsense built on server for aarch64 architecture?

dorodnic, 2020-03-03:
Hi @cimbar 
Publishing Jetson binaries is not fully automated, but we are trying to keep it reasonably up-to-date. We'll push 2.33.1 sometimes in near future. That said, librealsense releases are incremental so if 2.28 is working for you I would not recommend upgrading just yet. We are aware of several T265 issues and are still working on these.

cimbar, 2020-03-03:
As it is known, T265 can produce odometry messages with nan value and new releases have some fixes about this issue, so i would like to test it. I can also build it from source but i suffer some usb issues on new releases when i built it from source. Therefore, binary release of newer versions will be great and i am looking forward it, thanks

closed 2020-07-06 by dorodnic