realsenseai/librealsense #5960: Reset the T265 tracking with an initial transform?

antithing, 2020-03-03:
------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 

### Issue Description

When the T265 is launched and tracking is started, the pitch and roll are set from the imu, and the yaw is set to zero. The position is set to zero.
Is it possible to pass a position and yaw rotation as a start position?

For example, I want the tracking to start at a certain position, with a certain yaw rotation, instead of zero.

I am attempting this by simply adding a position vector and a quaternion to the pose data, the position works well, but i am having trouble with the quaternion.

Can I reset the tracking to start at a given transform?

Thanks.

MartyG-RealSense, 2020-03-03:
The ability to set a start position has been asked for before.  In June 2019, RealSense T265 expert @radfordi suggested a solution was being looked at, "allowing the user to define the forward direction in device's body coordinate system and then to define which direction in the world coordinates it could be lined up with initially. e.g. the "pointing out of the top of the camera is the initial forward direction" and it "should point in the world x direction initially".

https://github.com/IntelRealSense/librealsense/issues/4080#issuecomment-498111484

Somebody asked for an update on this potential feature last month.

https://github.com/IntelRealSense/librealsense/issues/4080#issuecomment-585827308

The intermediate approach for changing start position in the meantime was to have the T265 tilted during initilization.

MartyG-RealSense, 2020-03-03:
Regarding setting yaw: in June 2019, RealSense T265 expert @radfordi suggested a solution was being looked at, "allowing the user to define the forward direction in device's body coordinate system and then to define which direction in the world coordinates it could be lined up with initially. e.g. the "pointing out of the top of the camera is the initial forward direction" and it "should point in the world x direction initially".

https://github.com/IntelRealSense/librealsense/issues/4080#issuecomment-498111484

Somebody asked for an update on this potential feature last month.

https://github.com/IntelRealSense/librealsense/issues/4080#issuecomment-585827308

The intermediate approach for changing start position in the meantime was to have the T265 tilted during initilization.

antithing, 2020-03-03:
Hi, Thanks for getting back to me.

What I want seems slightly different to those links. I am trying to write a 'reset' function that will either restart the tracking relative to a world-zero position / rotation, or to add a position / rotation value on top of the tracking pose to give the same result.

Does adding a wheel-odometry calibration file have this result?  Eg: If the calibration file has a position offset of 10,10,10, will the tracking pose begin at -10,-10,-10? 
(We are not using odometry data, however).

Thanks again.

MartyG-RealSense, 2020-03-03:
Would manually resetting the 6DOF pose of the T265 achieve what you need, please?

https://github.com/IntelRealSense/librealsense/issues/3397

antithing, 2020-03-03:
It would, IF I could pass an initial transform in when resetting, to start the tracking at a specific position / orientation. Is this implemented?

MartyG-RealSense, 2020-03-03:
@radfordi or @ev-mp would be able to answer this question well.  Looking at the **rs-trajectory** example program though, which calculates the transformation matrix based on initial position and pose, I wonder if you could determine the output transformation to your own specification if you could substitute your own vectors for auto_q and auto_t:

[image]

https://dev.intelrealsense.com/docs/rs-trajectory

neilyoung, 2020-03-03:
If your aim is to start at any point in your room and have a valid world positioning, even though your cam always starts from 0, then check out the concept of set_/get_static_node. The retrieval of an arbitrary set static node allows you each time again to have this static node be the origin of your world coordinates.  This is done in a POC here https://neilyoung.serveblog.net/jekyll/update/2020/02/13/an-application-of-the-intel-realsense-t.265.html

The next days I will add an easy solution  to define the required reference point (static_node) on a map as well as to geo reference a floor plan, so that the display of the derived coordinates is done in realtime on a floor plan. 

This then will be a real turn key solution. Drop me a mail if you are interested.

smartynenko, 2020-03-28:
> Would manually resetting the 6DOF pose of the T265 achieve what you need, please?
> 
> #3397

I tried to manually reset pose, but it also resets tracking confidence and requires to move camera back and forth to regain confidence. Is it possible to reset w/o loosing tracking confidence?

RealSenseSupport, 2020-09-10:
Thank you for highlighting the need to manually reset pose, reset tracking and start backup with an initial transform. We have moved our focus to our next generation of products and consequently, we will not be addressing this feature in the T265.

closed 2020-09-15 by RealSenseSupport

DJSures, 2020-12-05:
Wait - so a product that is still for sale on your website AND promoted has actually been secretly discontinued? And the only way we find out is by reading a comment to an issue from September 15, 2020? Intel, if you expect us to support your sensors in our products, how do we know they will even be available in a month or two? Given that you don't go out of your way to tell anybody.