realsenseai/librealsense #5962: Remove ghost noise when camera close to target

szx0112, 2020-03-03:
It is known the d435 stereo camera has min distance of sensing. However, it seems that in our project, it is inevitable to face some conditions when some parts of the image is close to the object.

At the areas with close distance, d435/d435i senses sparse dot as 'noise' instead of totally a black background (As image below).

[image]

I am wondering if there is any build-in filter or solutions can help that we can remove/diminish those 'noise'?

Any comment is appreciate.

agrunnet, 2020-03-03:
Have you considered using the disparity shift parameter under advanced mode?

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

Hi @szx0112, Did you try changing the disparity shift parameter? Do you have further questions?
If we don't hear from you in 7 days, we will close this issue.

closed 2020-03-27 by RealSenseCustomerSupport