realsenseai/librealsense #5964: No depth details after 4m range from D435i camera

malith1992, 2020-03-04:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (05.12.02.100) | 
| Operating System & Version |   {Windows 10 & Ubuntu 18.04}  |                                         | 
| Platform                                 | PC and JetsonNano  |
| Language                            |  {realsense viewer}                          | 
| Segment			|  {Robot}                          | 

I have set Intel Realsense D435i camera nearly 5.5m above the ground. and I check the stereo view and RGB view from Realsense viewer. The objects far more than 4m from the camera are not detected in the stereo view. All of the pixels are black including the background. Simply there are no   depth details after 4m from camera. But the documentation of the camera says depth details are available upto 10m. I have attached the images from the output streams.  

**Is there any way to change the depth measuring range to 10m from the relasense viewer and the pyrealsense2 API ?**

[image]

[image]

MartyG-RealSense, 2020-03-04:
In the 'Post Processing' section of the RealSense Viewer's options side panel, please check if an option called 'Threshold Filter' is enabled (if it is on, there will be a blue colored icon beside it).  

This filter, when enabled, is set by default to exclude background detail further than 4 meters from the camera.  If it is enabled (blue), disabling it by left-clicking on the blue icon beside it should let the camera see to its full depth sending range.

malith1992, 2020-03-04:
> In the 'Post Processing' section of the RealSense Viewer's options side panel, please check if an option called 'Threshold Filter' is enabled (if it is on, there will be a blue colored icon beside [...]

It worked on realsense viewer. Thank you very much. Is there any way to turn it on and off or put a value for the Threshold filter using pyrealsense2 API on my python code.

MartyG-RealSense, 2020-03-04:
Post processing filters are inactive by default when writing your own application and have to be specifically turned on and off by your code.  Intel have a Python tutorial on post processing filters in the link below.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

In regard to a Python threshold filter, a RealSense community member created such a filter that was integrated into the SDK in March 2019.

https://github.com/IntelRealSense/librealsense/commit/84f78c9888a4e2ae48100e52df2e91c5ab82e683

I could not find an example script for its use though so I will ask one of the RealSense team on this forum.  @dorodnic

malith1992, 2020-03-04:
Thank you very much for the sources. I will refer them and apply it to my code.

closed 2020-03-04 by malith1992

MartyG-RealSense, 2020-03-05:
Hi again @malith1992   A RealSense community member who was trying to depth-clamp with a threshold filter in C# has posted screenshot images of the solution they wrote, so I will post the link here for other C# users to refer to.

https://support.intelrealsense.com/hc/en-us/community/posts/360039445773/comments/360010898533