realsenseai/librealsense #5965: RTAB-Map can't read intel D435

CluelessIT, 2020-03-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | v2.31.0 | 
| Operating System & Version |    Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | NVIDIA Jetson nano  |
| SDK Version                            |  Jetpack                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  3D mapping                       | 

### Issue Description
I am having a issue getting my RTAB-Map to detect my intel realsense D435 camera. I am using a Standalone version of RTAB-Map and already have realsense SDK2.0 installed. Not sure why the source option does not allow me to select the realsense camera. Any advice would be appreciated thanks! @matlabbe

dorodnic, 2020-03-04:
This link can give you a good starting point:
https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i
You can ignore everything related to IMU

CluelessIT, 2020-03-04:
Thanks for the tip! but I need to use RTAB-Map standalone without ROS. Using this link will make me use RVIZ.

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

@CluelessIT You issue seems more related to RTAB-MAP project. Maybe you can create one ticket in RTAB-MAP project to get more help. Thanks!

matlabbe, 2020-03-10:
This is indeed on rtabmap side. Close this issue, refer to this post instead https://github.com/introlab/rtabmap/issues/490#issuecomment-597168049

RealSenseCustomerSupport, 2020-03-11:
----------------------------------------------

@matlabbe Thanks for the help on this!
@CluelessIT Will close this accordingly. Thanks!

closed 2020-03-11 by RealSenseCustomerSupport