realsenseai/librealsense #5966: D400/T265 ubuntu 18.04 Libusb issue

scifiswapnil, 2020-03-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265/D415  | 
| Firmware Version                   | 5.11.6  | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |     5.3.0-40-generic                | 
| Platform                                 | Lenovo w451  |
| SDK Version                            |  2.0 build (2.33.1)                     | 
| Language                            |  C++/Python                          | 
| Segment			|  Robot                          | 

### Issue Description
On ubuntu 18.04 I am unable to build librealsense 2.0 build (2.33.1). I keep on getting error related to libusb-1.0.0-dev. 
`realsense-viewer: symbol lookup error: /usr/local/lib/librealsense2.so.2.33: undefined symbol: libusb_get_port_numbers`

I did try reinstalling libusb and also complied it from the source. But still facing the same issue.

MartyG-RealSense, 2020-03-06:
Another RealSense user experienced this error.  Their solution was "forcing CMake to find the right version of libusb".

https://github.com/IntelRealSense/librealsense/issues/4557

A different approach may be adding **-D USE_SYSTEM_LIBUSB=false** to your Cmake instructions so that the installation ignores the libusb version.

scifiswapnil, 2020-03-06:
[image]
I was able to build, but again on running realsense-viewer, i run into the same problem.

MartyG-RealSense, 2020-03-06:
I believe that the example and tool binaries should get installed into /user/local/bin when including these CMake flags during building Librealsense:

-DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true

wf-hahaha, 2020-03-28:
hello，Did you solve it? how to solve it？

wf-hahaha, 2020-03-30:
@scifiswapnil @MartyG-RealSense hello，how to solve it？realsense-viewer: symbol lookup error: /opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.33: undefined symbol: libusb_get_port_numbers

scifiswapnil, 2020-03-30:
@wf-hahaha currently don't have the hardware with me. Lastly I checked, I was able to run the realsense utilities from the build folder only, but not after installing. Further to this I was looking for a way to bind the build folder to ROS-wrapper, so I can use the ROS wrapper

scifiswapnil, 2020-05-15:
As noted before " I was able to run the real sense utilities from the build folder only, but not after installing from /usr/local/bin". 
So for ROS-realsense2 wrapper, in the CMakeLists.txt in realsense2_camera I manually entered the path to the build directory 
```find_package(realsense2 enter_linrealsense_version PATHS path/to/build/directory)```
A temporary fix, but worked. Still have no clues about why does the executable after installing the build.

RealSenseSupport, 2020-09-21:
Thank you for highlighting the issue outlined regarding T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2024-04-01 by MartyG-RealSense