realsenseai/librealsense #5972: null pointer passed for argument "buffer" still present in 2.33.1

neilyoung, 2020-03-04:
Just wanted to remind you. Already raised several times.

Happens _sometimes_ when a map shall be saved.

MartyG-RealSense, 2020-03-07:
I will label this post as 'bug' so that it can be tracked by the RealSense team and investigated regarding whether it is a bug.

neilyoung, 2020-03-07:
Thanks @MartyG-RealSense. This problem is really a bit annoying. Imagine, you are walking through an area and intend to save the map afterwards. It is a 50:50 chance that it will fail and all the efforts are lost. More than that: You need to unplug the camera and restart from scratch, when it happens...

neilyoung, 2020-03-15:
Any development here?

MartyG-RealSense, 2020-03-15:
If this is a T265 related issue, the best way to get an answer will likely be to close this case and repost the exact same question as a new issue but make sure that the word 'T265' is included in the message title this time.  This should make sure the right person picks it up for answering.

neilyoung, 2020-03-15:
Thanks. Will do. Another bug ("Bus Error 10") today also popped up again

closed 2020-03-15 by neilyoung